solvespace/solvespace #1244: Segmentation fault on language change

ktrace, 2022-05-14:
### System information

- **SolveSpace version:** 3.1~0e0b0252
- **Operating system:** gentoo/Calculate

### Expected behavior
Try to change language:
Help -> Language -> (any language include English (US))

### Actual behavior

Segmentation fault

### Additional information

Try under strace and choose strace ended with:

openat(AT_FDCWD, "/usr/share/solvespace/locales/es_AR.po", O_RDONLY) = 15
newfstatat(15, "", {st_mode=S_IFREG|0644, st_size=59103, ...}, AT_EMPTY_PATH) = 0
newfstatat(15, "", {st_mode=S_IFREG|0644, st_size=59103, ...}, AT_EMPTY_PATH) = 0
lseek(15, 57344, SEEK_SET)              = 57344
read(15, "\"\nmsgid \"&OK\"\nmsgstr \"&Aceptar\"\n"..., 1759) = 1759
lseek(15, 0, SEEK_SET)                  = 0
read(15, "# Spanish/Argentina translations"..., 57344) = 57344
read(15, "\"\nmsgid \"&OK\"\nmsgstr \"&Aceptar\"\n"..., 4096) = 1759
close(15)                               = 0
write(2, "Missing (empty) translation for "..., 53) = 53
write(2, "\n", 1)                       = 1
write(2, "Missing (empty) translation for "..., 64) = 64
write(2, "\n", 1)                       = 1
write(2, "Missing (empty) translation for "..., 53) = 53
write(2, "\n", 1)                       = 1
write(2, "Missing (empty) translation for "..., 58) = 58
write(2, "\n", 1)                       = 1
write(2, "Missing (empty) translation for "..., 54) = 54
write(2, "\n", 1)                       = 1
--- SIGSEGV {si_signo=SIGSEGV, si_code=SI_KERNEL, si_addr=NULL} ---
+++ killed by SIGSEGV +++

bt in gdb:

Thread 1 "solvespace" received signal SIGSEGV, Segmentation fault.
0x000055555562c86c in ?? ()
(gdb) bt
#0  0x000055555562c86c in  ()
#1  0x00007ffff7d3f55e in Gtk::MenuItem_Class::activate_callback(_GtkMenuItem*) () at /usr/lib64/libgtkmm-3.0.so.1
#2  0x00007ffff6ec85ff in g_closure_invoke () at /usr/lib64/libgobject-2.0.so.0
#3  0x00007ffff6ed9dd5 in  () at /usr/lib64/libgobject-2.0.so.0
#4  0x00007ffff6ee0591 in g_signal_emit_valist () at /usr/lib64/libgobject-2.0.so.0
#5  0x00007ffff6ee0b3f in g_signal_emit () at /usr/lib64/libgobject-2.0.so.0
#6  0x00007ffff746f9f4 in gtk_widget_activate () at /usr/lib64/libgtk-3.so.0
#7  0x00007ffff7336e56 in gtk_menu_shell_activate_item () at /usr/lib64/libgtk-3.so.0
#8  0x00007ffff7337133 in  () at /usr/lib64/libgtk-3.so.0
#9  0x00007ffff74c2ed4 in  () at /usr/lib64/libgtk-3.so.0
#10 0x00007ffff6ec8829 in  () at /usr/lib64/libgobject-2.0.so.0
#11 0x00007ffff6edfc7b in g_signal_emit_valist () at /usr/lib64/libgobject-2.0.so.0
#12 0x00007ffff6ee0b3f in g_signal_emit () at /usr/lib64/libgobject-2.0.so.0
#13 0x00007ffff746cb64 in  () at /usr/lib64/libgtk-3.so.0
#14 0x00007ffff73218d0 in  () at /usr/lib64/libgtk-3.so.0
#15 0x00007ffff732332d in gtk_main_do_event () at /usr/lib64/libgtk-3.so.0
#16 0x00007ffff7014565 in  () at /usr/lib64/libgdk-3.so.0
#17 0x00007ffff7048062 in  () at /usr/lib64/libgdk-3.so.0
#18 0x00007ffff6dd3a0b in g_main_context_dispatch () at /usr/lib64/libglib-2.0.so.0
#19 0x00007ffff6dd3cb8 in  () at /usr/lib64/libglib-2.0.so.0
#20 0x00007ffff6dd3fa3 in g_main_loop_run () at /usr/lib64/libglib-2.0.so.0
#21 0x00007ffff73224f5 in gtk_main () at /usr/lib64/libgtk-3.so.0
#22 0x00005555555913d6 in  ()
#23 0x00007ffff66601ca in  () at /lib64/libc.so.6
#24 0x00007ffff6660278 in __libc_start_main () at /lib64/libc.so.6
#25 0x0000555555596221 in  ()

phkahler, 2022-05-14:
My builds on Fedora can not reproduce this. Can anyone else confirm it?

phkahler, 2022-06-01:
Can not reproduce.

ktrace, 2022-06-17:
Look like bug present only with dynamic mimalloc link (in gentoo's portage).

closed 2023-09-20 by phkahler