solvespace/solvespace #1246: Feature request: Faces Perpendicular Constraints

phkahler, 2022-05-16:
## Description

Like the title says. Allow the user to select two faces and apply a perpendicular constraint. This would be most useful for revolve groups to set the angle to 90 degrees. There are probably other uses as well.

@jwesthues  On implementation would it make sense to add faces to EntityBase::HasVector  and other minor tweaks so the existing perpendicular constraint can be applied, or should I make a new FACES_PERPENDICULAR constraint?

The problem I see wtih adding them to HasVector is that the vector would be the normal - perpendicular to the face. This would get confusing if someone tried to do a line and face perpendicular where they really want the line parallel to the face normal. The semantics of the math are different to the way we talk about it when it comes to faces.

jwesthues, 2022-05-16:
The feature makes sense to me, and implementation with `::HasVector()` and the existing constraints would do what you want for face-face angles with very little additional code.

Line-face angle is fundamentally a different thing, as you note. For example, line-line and face-face perpendicular both constrain 1 DOF, but line-face perpendicular constrains 2 DOF. I guess you could swap the definitions of parallel and perpendicular when it's line-face, but that might be confusing too (and what would you do with angle dimensions?). It might be better just to document the behavior, or even warn the user each time when such a constraint is created.

phkahler, 2022-06-02:
Oops, this is a duplicate of #1115, but I'll close that one.

closed 2022-06-12 by phkahler