solvespace/solvespace #1247: SolveSpace fails to solve solvable constraints

tc7000, 2022-05-20:
### System information

- **SolveSpace version:** 3.0~ab00823a
- **Operating system:** Ubuntu 20.04.4 LTS 64-bit

### Expected behavior

There should be no error solving constraints.

### Actual behavior

unsolvable constraints error (distance and angle)

### Additional information

Extract attached zip file, open test.slvs, then modify the angle constraint to 60°. Change again the angle to 30°.
[test.slvs.zip](https://github.com/solvespace/solvespace/files/8741298/test.slvs.zip)

phkahler, 2022-05-26:
@tc7000  Thank you for such a simple test case.

That's really annoying. It can handle a jump from 60 to 40, but not 60 to 30. This seems similar to another solver failure issue but also much simpler and more likely for people to run into.

ruevs, 2022-06-03:
Nice test case! Changing the angle from 60 to anything between 27-32 causes the solve to fail.
59.1 -> 28. fails as well, 59.0 -> 28.0 does not.  Interesting.

phkahler, 2022-06-03:
This is among the things #1149 might address. Need a mathematician for this stuff  ;-)

ruevs, 2022-06-03:
There is some nice debug code from @Evil-Spirit here https://github.com/Evil-Spirit/solvespace-master/commits/solver-debug I have not pulled it into master in years but it is a rather interesting tool. No time now but I'll do it... some day.

ruevs, 2022-06-06:
I rebased
https://github.com/Evil-Spirit/solvespace-master/commits/solver-debug
onto master here:
https://github.com/ruevs/solvespace/tree/EvilSpirit_solver-debug
it is interesting to watch the solve fail, and see the equations but no immediate help. I need to think/debug much more...

ruevs, 2022-07-08:
Issue https://github.com/solvespace/solvespace/issues/21 seems to me somewhat related. Even though the details differ.

zwn, 2024-02-07:
I have also a somewhat simple reproduction case:
- create a block (extruded rectangle) and save file
- create a new file to be used as assembly
- imported the block into the assembly
- constrain the orientation of the imported block (_constrain normals in same orientation_) on the Z axes - works as expected
- undo the constrain
- import the same block for a second time
- try to add the same constrain that worked before on the firs block - unsolvable
- try to add the same constrain on the newly imported block - works as expected

[unsolvable.zip](https://github.com/solvespace/solvespace/files/14201184/unsolvable.zip)

closed 2026-10-08 by phkahler