solvespace/solvespace #1249: Impossible to export curves as curves, not series of straight lines

sokol07, 2022-05-27:
### System information

- **SolveSpace version: 3.0~0e0b0252** 
- **Operating system: Windows 10** 

### Expected behavior

I was trying to export a shape to DXF file to process it in FreeCAD as a technical drawing for manufacturing. I want to have all circles exported as real circles so that I can select them as circles in FreeCAD TechDraw module and add diameters, etc.

### Actual behavior

I exported a 3D wiremesh (I also tried 2D view with the same result) to a DXF file, imported it into FreeCAD and all cirles in the project were converted into series of tiny straight lines.

### Additional information

I have the "export curves as piecewise linear" option turned off.
Similar problem to: https://solvespace.com/forum.pl?action=viewthread&parent=2298&tt=1524026523
Zip file of my project below.

[gora.zip](https://github.com/solvespace/solvespace/files/8786757/gora.zip)

rpavlik, 2022-07-06:
So I've exported models as surfaces and imported into FreeCAD, and I get curves there. I have not. tried DXF. That said I was pretty sure I had exported to PDF and gotten curves, even if not ideal. Do you have "convert to tri mesh" somewhere in your model maybe?

ruevs, 2022-09-14:
https://solvespace.com/forum.pl?action=viewthread&parent=4799&tt=1663184291

rpavlik, 2022-09-14:
Ok so it's probably a DXF-specific limitation.

rpavlik, 2022-09-14:
Step also exports curves