solvespace/solvespace #1253: variables

ghsp22, 2022-06-10:
#Idea or Enhancement

I noticed sometimes I have to restart constructing. Or I have collected several measurements that go down as distance constraints. 
Also it seems even in the tutorial sometimes the same measure value is inserted several times.

Maybe it would be worthwhile to allow the user to enter a list of variables and later use variable names in the distance field.
Ideally even using a cvs (or other file) with the variable name; value to be important at start and fill in the variables.

ruevs, 2022-06-11:
This is a duplicate and is one of the oldest feature requests:

- Parametric sketches, parametric constraints, variables. inequality constraints etc. </br>Examples: `x=2` `y=3` `z=x+5` `3*x+z` `x`  `<7`</br>https://github.com/solvespace/solvespace/issues/77 https://github.com/solvespace/solvespace/issues/81   https://github.com/solvespace/solvespace/issues/422 https://github.com/solvespace/solvespace/issues/80 https://github.com/solvespace/solvespace/issues/166 (duplicates https://github.com/solvespace/solvespace/issues/273 https://github.com/solvespace/solvespace/issues/542 https://github.com/solvespace/solvespace/issues/931 https://github.com/solvespace/solvespace/issues/1030)

There is a pinned issue https://github.com/solvespace/solvespace/issues/1024 that has links to many common requests and problems.

closed 2022-06-11 by ruevs

Kolostov, 2024-10-14:
A list bowser where name=1 and expression execution on all values within the sketch is a great idea.

I want to sketch a mechanised window frame but windows are different size (w,h) so most of my elements must be constrained to these variables.

Another thing in my issue is to set thickness of material as a variable, so if that changes then many of elemnets has to change too.