solvespace/solvespace #1261: SolveSpace 3.1 is crashing so frequently without any warning/error on assembling parts.

simrim1, 2022-06-24:
### System information

SolveSpace version: 3.1~70bde63c
Operating system: Windows 10

Expected behavior

I am designing a CNC machine in SolveSpace and trying to combine (assemble) different parts (SolveSpace sub assembly files) in a separate file which has its own drawing as well. 
It should work as normal and should not crash.

Actual behavior

It keeps crashing and has become almost unusable.

Additional information

The files sub assemblies which further consists of sub assemblies are larger than usual files. The final assembly has become of size 45MB. It looks like memory leak issue. The application crashes silently without showing any error.

ruevs, 2022-06-25:
@simrim1 would you be willing to share your model? If not publicly et least privately?
Does it crash with 3.0?
Are you by any chance the author of this video https://www.youtube.com/watch?v=iuBfzpcCO5c ?

phkahler, 2022-06-25:
I'm wondering if this is a new issue or if it's the size of the sketch/assembly running into a limitation.

We have seen cases where the performance increases allowed people to hit limitations before.

So yeah, is this unique to version 3.1 or does it happen with 3.0 as well? Assuming the files are compatible (no 3.1 specific features).

simrim1, 2022-06-25:
@ruevs Spot on man. Your searching capabilities are better than Google. I am really curious how you related my post to my video on Youtube.
Coming back to the issue, I have not tested it in 3.0 and I don't think I can because I have made all my designs in 3.1 taking benefit of better Helix feature in 3.1.

I can send you files privately. I am going to public them all in my next 2-3 videos. I am keeping them private for the time being. Please let me know how to send you files.

@phkahler I guess this is memory leak issue when it comes to assembly but I am not sure. In the final assembly when we have done 2-4 operations doesn't matter it is saving the file, moving something with mouse. applying any constraint or something else which needs a geometry calculation, it crashes.

ruevs, 2022-06-25:
> Please let me know how to send you files.

Email "dpr" at the domain you can see in my profile.

ruevs, 2022-06-26:
@simrim1 First thank you for sending me your CNC lathe/router model! Very impressive job! It is by far the most complicated SolveSpace project I have seen.

I found your YouTube channel a few months ago when you posted your ["Resolving Issues of Solvespace on Windows 7 and Windows XP."](https://www.youtube.com/watch?v=C6g2F-nNXK4) video and linked to my [Windows XP build](https://github.com/solvespace/solvespace/issues/1036#issuecomment-1019377777). And I stumbled upon your [new video](https://www.youtube.com/watch?v=iuBfzpcCO5c) by searching for recent SolveSpace content on YouTube just two days after you had posted it. I even posted it on the SolveSpace forum https://solvespace.com/forum.pl?action=viewthread&parent=4653&tt=1656063760 And then you opened this bug :-)

I see that you [have one more now](https://www.youtube.com/watch?v=2EZxi9tPfws).

ruevs, 2022-06-26:
And now to the problem :-)

It is not necessarily a memory leak. It simply runs out of it. Probably through [heap fragmentation](https://www.design-reuse.com/articles/25090/dynamic-memory-allocation-fragmentation-c.html).

At about 1.2GB allocated after opening your MyCNCBaseAssembly.slvs and toggling back and forth through the groups a few times. The failure is eventually at this memory allocation:
https://github.com/solvespace/solvespace/blob/c20c4dab30e86c103678daeb2d936c1407b3cbb8/src/dsc.h#L265

It tries to allocate 131072 objects of type `List<SOutline> l;` the calling code is here:
https://github.com/solvespace/solvespace/blob/c20c4dab30e86c103678daeb2d936c1407b3cbb8/src/mesh.cpp#L1119

Here is the `SOutline` class
https://github.com/solvespace/solvespace/blob/c20c4dab30e86c103678daeb2d936c1407b3cbb8/src/polygon.h#L307

The size of `SOutline` is 100 bytes so it is trying to allocate roughly 13107200 bytes or 13MB.

At 1.2GB of RAM used and a 32 bit process being able to allocate about 2GB on a modern Windows I think that the problem may be heap fragmentation rather than actually running out of heap.

The problem (in this case) is fixed simply by compiling with the [/LARGEADDRESSAWARE](https://docs.microsoft.com/en-us/cpp/build/reference/largeaddressaware-handle-large-addresses) linker option, which allows 32 bit processes to access more than 2GB of RAM. I will attach such a version for you to test in a few minutes. And I will also make a pull request - this has been discussed before https://github.com/solvespace/solvespace/pull/968#issuecomment-906206212.

Also SolveSpace compiled as 64bit does not have the problem.

It is amazing (and very good by itself) that it is at all possible to run out of RAM - [up to about six months ago this was really hard and before 3.0 it was practically impossible](https://github.com/solvespace/solvespace/issues/1186) - it would become unacceptably slow long before running out of memory.

Here is the call stack for full context.

```
 	solvespace.exe!SolveSpace::List<SolveSpace::SOutline>::ReserveMore(int howMuch) Line 265	C++
 	solvespace.exe!SolveSpace::List<SolveSpace::SOutline>::AllocForOneMore() Line 279	C++
 	solvespace.exe!SolveSpace::List<SolveSpace::SOutline>::Add(const SolveSpace::SOutline * t) Line 283	C++
>	solvespace.exe!SolveSpace::SOutlineList::AddEdge(SolveSpace::Vector a, SolveSpace::Vector b, SolveSpace::Vector nl, SolveSpace::Vector nr, int tag) Line 1120	C++
 	solvespace.exe!SolveSpace::PolylineBuilder::GenerateOutlines::__l2::<lambda>(SolveSpace::PolylineBuilder::Vertex * next, SolveSpace::PolylineBuilder::Edge * e) Line 234	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::PolylineBuilder::Generate(const std::function<void __cdecl(SolveSpace::PolylineBuilder::Vertex *,SolveSpace::PolylineBuilder::Vertex *,SolveSpace::PolylineBuilder::Edge *)> & startFunc, const std::function<void __cdecl(SolveSpace::PolylineBuilder::Vertex *,SolveSpace::PolylineBuilder::Edge *)> & nextFunc, const std::function<void __cdecl(SolveSpace::PolylineBuilder::Edge *)> & aloneFunc, const std::function<void __cdecl(void)> & endFunc) Line 170	C++
 	solvespace.exe!SolveSpace::PolylineBuilder::GenerateOutlines(SolveSpace::SOutlineList * sol) Line 242	C++
 	solvespace.exe!SolveSpace::Group::GenerateDisplayItems() Line 506	C++
 	solvespace.exe!SolveSpace::Group::Draw(SolveSpace::Canvas * canvas) Line 652	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::DrawPersistent(SolveSpace::Canvas * canvas) Line 682	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::Draw(SolveSpace::Canvas * canvas) Line 722	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::Paint() Line 886	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::Platform::WindowImplWin32::WndProc(HWND__ * h, unsigned int msg, unsigned int wParam, long lParam) Line 775	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::Platform::RunGui() Line 1711	C++
 	solvespace.exe!main(int argc, char * * argv) Line 29	C++
 	solvespace.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 40	C++
```

ruevs, 2022-06-26:
@simrim1 Here is a large address aware 32 bit Windows build - OpenGL 3, multi threaded (OpenMP).
[SolveSpace_Win32_x86_Post3.1_c20c4dab_LARGEADDRESSAWARE.zip](https://github.com/solvespace/solvespace/files/8986464/SolveSpace_Win32_x86_Post3.1_c20c4dab_LARGEADDRESSAWARE.zip)

Please try it and let me know if it crashes or not.

And since there is a YouTube video with your project already I hope you do not mind that I share a screen shot of your model for others to see. It is really impressive.
[image]

And yes, the X axis does not quite align with the nut ;-)

ruevs, 2022-06-26:
And here are the memory limits under different conditions on Windows:
https://docs.microsoft.com/en-us/windows/win32/memory/memory-limits-for-windows-releases?redirectedfrom=MSDN

Memory type | Limit on X86 | Limit in 64-bit Windows
-- | -- | --
User-mode virtual address space for each 32-bit process | 2 GB<br/>Up to 3 GB with IMAGE_FILE_LARGE_ADDRESS_AWARE and 4GT | 2 GB with IMAGE_FILE_LARGE_ADDRESS_AWARE cleared (default)<br/>4 GB with IMAGE_FILE_LARGE_ADDRESS_AWARE set

phkahler, 2022-06-26:
@ruevs and @simrim1  The question that always comes to my mind with complex designs:  Did you have to force it to triangle mesh?

ruevs, 2022-06-26:
@phkahler not a single one from a total of 37 models and assemblies is forced to triangle mesh.

There are errors (naked edges) in only two assemblies of longer screws made from shorter screws (helixes)
MyCNCLeadScrew8mm500Assembly.slvs
MyCNCLeadScrew8mm600Assembly.slvs

Everything else looks good.

simrim1, 2022-06-26:
@ruevs Thanks a million for having a look at it. I am sorry I was away. Just switched on my computer and found your messages. You response if very detailed. Although I am a programmer myself, most of the things you explained are out of my domain but some of them I understand.

Where is 64bit version by the way?

I have tested your newly built binary you attached here. It is still crashing. All I did was to move the gantry back and forth about 4-6 times and I believe either Autosave or some computations after movement made it crash.

Another thing I noticed is when I open the main assembly initially. I check the RAM usage in Task Manager and is 883-885MB. When I move the gantry it stats going up gradually which I think, is saving its previous state for Undo/Redo purposes or for something else. So every operation takes up some memory. The strange thing is, before crashing, its RAM comes down and then it crashes.

As for sharing the drawing, you can share any drawing as image from these files. They are kept private for the time being, so people try to learn SolveSpace. I had recorded initial videos as tutorials/guides for people in subcontinent with a very low English language skills and they looking something like this in their own language. I will make them public soon anyway.

I didn't see any errors by the way in LeadScrew assemblies in the browser window. Will you please point me to correct location in SolveSpace when I can see these errors.

One question though, Is SolveSpace not for complicated designs?

Thanks

simrim1, 2022-06-26:
@ruevs O yes, I forgot to address your last comment about the alignment of Nut in "X-Axis". This is actually Y-Axis in my design. I was actually pointing it out in the video and how to find out how much misaligned it is.

Thanks again for going through the video and noting the point.

phkahler, 2022-06-30:
> I didn't see any errors by the way in LeadScrew assemblies in the browser window. Will you please point me to correct location in SolveSpace where I can see these errors.

@simrim1  Under Analyze->Show Naked Edges
Also for assemblies Analyze->Show Interfering Parts

The second one is very likely to fail on the lead screw since it goes through a nut and the Helical surfaces are not perfect. I have not tried this.

Making the long screws one piece might help if I read @ruevs comment correctly.

> One question though, Is SolveSpace not for complicated designs?

It is for whatever you want, however the performance degrades as models become more complex, sometime so much that people give up. We keep making performance improvements and people keep pushing it farther. I don't think your design is *that* complex by the number of parts, but those screws will slow it down a bit ;-)

A similar situation exists with the NURBS surface bugs. The more complex a design the more likely you are to encounter issues, but we keep fixing things and people keep designing ever more complex things until they can't.

ruevs, 2022-07-01:
> @simrim1 Under Analyze->Show Naked Edges
> Also for assemblies Analyze->Show Interfering Parts

Indeed on the final assembly they show many red edges.

From the individual parts only the two long screws I mentioned above have NURBS problems. And @phkahler is right - if you model them as two individual long screws instead of assembles of the short cestion their NURBS problems (red edges where the sections meet) should disappear.

rpavlik, 2022-07-01:
I don't think it require a debug build, just an option in CMake. I've had it on for a long time "by accident", finally turned it off recently.

simrim1, 2022-07-02:
@phkahler @ruevs Thanks for pointing it out how to find out issues in drawing. It is right that screw issue cannot be resolved unless you keep it at a very precise location so that leadscrew threads don't intersect with nut threads.

@ruevs I still believe it is memory leak issue. If you open a complex design (less than the main assembly) and some operations and then without closing solvesoace open the main assembly and it will crash. It suggests that memory leak had already happened.

If there is a memory leak, is there way to find out? If yes, then following steps could help to spit it.

1. Open a fresh instance of solvesoace.
2. Check the memory status.
3. Open the main assembly and do 4-5 operations like moving gantry back and forth.
4. Save the file.
5. Open a new drawing with File->New.
6. Check the memory status and compare it with the memory status at step 2.

If there is a memory leak it will be spotted.

I will design a long leadscrew in a single file. As far as I can remember my computer got stuck with 500mm leadscrew. it could be in the version 3.0. I will give it a try with a single long leadscrew in 3.1.

I have not tried it with Linux yet. This issue doesn't exist in Linux?

Thanks

ruevs, 2022-07-02:
> This could still be a memory leak right? CI doesn't cover everything. Does ASAN require a debug build, that would be painfully slow but may be worth a try.

It could - I ran it now with the address sanitizer of VS 2019 enabled. It did not catch anything.

ruevs, 2022-07-02:
@simrim1 here is a x64 version for your experiments.
[SolveSpace_Win32_x64_Post3.1_d6e1b230.zip](https://github.com/solvespace/solvespace/files/9033708/SolveSpace_Win32_x64_Post3.1_d6e1b230.zip)

For now I can not catch a memory leak.

phkahler, 2022-07-04:
@ruevs and maybe @rpavlik how much memory does solvespace eat for undo/redo history. I thought it was very minimal, just working on the requests. But does it - perhaps accidentally - keep multiple meshes and shells?

simrim1, 2022-07-04:
> @simrim1 here is a x64 version for your experiments. [SolveSpace_Win32_x64_Post3.1_d6e1b230.zip](https://github.com/solvespace/solvespace/files/9033708/SolveSpace_Win32_x64_Post3.1_d6e1b230.zip)
> 
> For now I can not catch a memory leak. But I can observe memory usage going slowly up after opening and closing your assembly many times in the same session.

@ruevs Thanks for the 64bit binary. It worked very well. Did a number of operations and SolveSpace didn't crash. Is it not a good idea to include 64bit binary in the assets?

The memory usage going up is what I was referring to. When you open complex design and do some operations and then open a new design without closing Solvespace going to File->New it should show the memory status as if you have just opened a new instance of SolveSpace. But if it still captures more than that, it means the memory is being eaten by something in the SolveSpace which it should have freed now. Am I right?

ruevs, 2024-10-26:
Attached is a x64 OpenMP (multi core) build of SolveSpace at c619fa9
[SolveSpace_x64_OpenMP_ruevs_c619fa9.zip](https://github.com/user-attachments/files/17529343/SolveSpace_x64_OpenMP_ruevs_c619fa9.zip)

closed 2025-06-21 by ruevs

ruevs, 2025-06-21:
By popular demand the CI now builds 64bit binaries on Windows.
This allows SolveSpace to open large models (by accessing >4GB RAM).
https://github.com/solvespace/solvespace/pull/1604

As always you can get the latest binaries from here:
https://github.com/solvespace/solvespace?tab=readme-ov-file#via-automated-edge-builds

PR #1262 (2022-06-30): Win32: Allow 32 bit SolveSpace to access up to 4GB of RAM.
Link 32 bit SolveSpace for Windows with /LARGEADDRESSAWARE which allows
it to access up to 3GB of RAM on a properly configured 32 bit Windows and
up to 4GB on 64 bit.

See: https://msdn.microsoft.com/en-us/library/aa366778
https://docs.microsoft.com/en-us/cpp/build/reference/largeaddressaware-handle-large-addresses
https://docs.microsoft.com/en-us/windows/win32/debug/pe-format#characteristics
https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-server-2003/cc786709
https://docs.microsoft.com/en-us/windows/win32/mem [... cut]