solvespace/solvespace #1263: Crash (assert) on load file - regression?

rpavlik, 2022-06-30:
### System information

- **SolveSpace version:** git bisect says it appeared in 465506670ceaa1f2004b7aa8d5979c9622b3d565 @phkahler 
- **Operating system:** Windows 10, Debian Bullseye

### Expected behavior

The attached file should open and be able to export to STL for 3d printing.

### Actual behavior

Crash/assertion failure:

```
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
File ./src/dsc.h, line 526, function FindById:
Assertion failed: t != nullptr.
Message: Cannot find handle.
Aborted
```

### Additional information
[wheel sleeve2.zip](https://github.com/solvespace/solvespace/files/9021231/wheel.sleeve2.zip)

ruevs, 2022-06-30:
@phkahler The pull request for your (and mine) bug fixes is this one https://github.com/solvespace/solvespace/pull/1201. The original one that implemented the feature is https://github.com/solvespace/solvespace/pull/1054 I found it through my commit message here https://github.com/solvespace/solvespace/commit/026f9369894bd

So the overall story so far is:
[A workplane can now be defined from a workplane defined by point and …](https://github.com/solvespace/solvespace/commit/026f9369894bd526e049ee0fbf2ee928ffdde612)
@[ruevs](https://github.com/solvespace/solvespace/commits?author=ruevs) authored and phkahler committed on Feb 3, 2022

[Fix normal issue, when a workplane is created from a point and normal…](https://github.com/solvespace/solvespace/commit/465506670ceaa1f2004b7aa8d5979c9622b3d565)
@[phkahler](https://github.com/solvespace/solvespace/commits?author=phkahler) committed on Feb 3, 2022

[Fix "Sketch in New Workplane" point & normal to set correct orientati…](https://github.com/solvespace/solvespace/commit/4308dc136b03d6727da96ec75cc7ae27d169040a)
@[MaxiPaille ](https://github.com/solvespace/solvespace/commits?author=MaxiPaille ) authored and phkahler committed on Jun 27, 2021

[Improve "Sketch in New Workplane" by adding two way of construction](https://github.com/solvespace/solvespace/commit/3ccf7845f5271adb9dc097a1cac4f7cfe54f88b3)
@[MaxiPaille ](https://github.com/solvespace/solvespace/commits?author=MaxiPaille ) authored and phkahler committed on Jun 27, 2021

ruevs, 2022-06-30:
It asserts here: https://github.com/solvespace/solvespace/blob/d6e1b23006580fb6bfb8d562264030524993bf1b/src/group.cpp#L466 this code was introduced here https://github.com/solvespace/solvespace/commit/465506670ceaa1f2004b7aa8d5979c9622b3d565 on February 2, 2022.

This means that any models using "WORKPLANE_BY_POINT_NORMAL" and created with a version after [**_Jun 27, 2021_**]( https://github.com/solvespace/solvespace/commit/3ccf7845f5271adb9dc097a1cac4f7cfe54f88b3) and before [**_February 2, 2022_**](https://github.com/solvespace/solvespace/commit/465506670ceaa1f2004b7aa8d5979c9622b3d565) will crash versions compiled after the latter commit.

Hopefully there are very few such models "in the wild" since we have not done official releases between these two dates.

@phkahler I think I thought about this incompatibility we will be creating when you and I [discussed it](https://github.com/solvespace/solvespace/issues/1120#issuecomment-1025221510) but it seems I did not mention it :-(

rpavlik, 2022-06-30:
Ohh, hmm. Didn't realize I was using a new feature at the time :)  Wonder if it's worth adding compat/conversion code. I was just surprised when I loaded a model I made recently in 3.1 and it went boom - apparently I was using an older build still.

rpavlik, 2022-06-30:
Actually maybe a better idea - is there a way to partially fail loading, such that previous groups are intact but there's a final "file corrupted" group when a problem is hit, instead of an assert? Maybe more sustainable than migration support.

phkahler, 2022-06-30:
> Wonder if it's worth adding compat/conversion code.

I was thinking about this on my way home. For your file you should be able to manually (yeah I know) add the workplane handle to the failing group in predef.entityB and it should load. The problem is figuring out which workplane to take the handle from, which makes automating it a bit difficult. I suppose the most recent sketch-in-workplane group would be a good guess?

Another option would be to check if predef.entityB is zero (if that's what it defaults to) and substitute the other behavior, which would allow sketches "broken" this way to load but they would still exhibit the bug that was fixed (stuff not following the normal if it's rotated).

phkahler, 2022-06-30:
> Actually maybe a better idea - is there a way to partially fail loading, such that previous groups are intact but there's a final "file corrupted" group when a problem is hit, instead of an assert? Maybe more sustainable than migration support.

This is an interesting idea. Load as far as possible and then display a message indicating incomplete loading due to an error? I'm assuming you'd want this to be some kind of generic error trapping so it would work in other cases in the future? I'm not an error trapping kind of coder myself, but I did something like this last millennium for a file format incompatibility. I just don't remember how.

rpavlik, 2022-07-01:
Yeah - I don't mind revising my model to not be broken, but it's nicer if I could load and reuse at least some of it.

ruevs, 2022-08-02:
There is at least one mode such model "in the wild"
https://solvespace.com/forum.pl?action=viewthread&parent=4690
I'm attaching it here for completeness:
[plug_blank.zip](https://github.com/solvespace/solvespace/files/9242289/plug_blank.zip)