solvespace/solvespace #1264: Regression blocking Debian

rpavlik, 2022-06-30:
### System information

- **SolveSpace version:** 3.1~rc1
- **Operating system:** Debian testing

### Expected behavior
autopkgtests run normally

### Actual behavior

"no surfaces to export" on s390x: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1013163

```
$ solvespace-cli export-surfaces --output /tmp/autopkgtest-lxc.qokm0ro_/downtmp/autopkgtest_tmp/%_surfaces.step ./test/group/translate_asy/normal_v22.slvs
Missing (absent) translation for group-name'#references'
Missing (absent) translation for group-name'sketch-in-plane'
Error: The model does not contain any surfaces to export.
Written '/tmp/autopkgtest-lxc.qokm0ro_/downtmp/autopkgtest_tmp/./test/group/translate_asy/normal_v22_surfaces.step'.
```

and

```
solvespace-cli export-surfaces --output /tmp/autopkgtest-lxc.qokm0ro_/downtmp/autopkgtest_tmp/%_surfaces.step ./test/group/translate_nd/normal_v22.slvs
Missing (absent) translation for group-name'#references'
Missing (absent) translation for group-name'sketch-in-plane'
Generate::ALL took 44 ms
Error: The model does not contain any surfaces to export.
Written '/tmp/autopkgtest-lxc.qokm0ro_/downtmp/autopkgtest_tmp/./test/group/translate_nd/normal_v22_surfaces.step'.
```
This did work with the previous Debian package version, which is a 3.0RC.

ruevs, 2022-06-30:
@rpavlik So is this is a custom Debian test suite/script using the files from our test suite and solvespace-cli ?

I am not on Linux but I did run the equivalent of the last command

```
solvespace-cli export-surfaces --output 
/tmp/autopkgtest-lxc.qokm0ro_/downtmp/autopkgtest_tmp/%_surfaces.step 
./test/group/translate_asy/normal_v22.slvs
```

On Windows:

```
[C:\Users\xxxx\Desktop\solvespace\buildmsvc\bin\Debug]solvespace-cli export-surfaces --output .\surfaces.step ..\..\..\git\test\group\translate_nd\normal_v22.slvs
```
and it exported a valid step file. This was in debug mode. Here is the output:

```
Missing (absent) translation for group-name'#references'
Missing (absent) translation for group-name'sketch-in-plane'
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 03808000
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 03808000 was valid after all)
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 03808000
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 03808000 was valid after all)
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 03810000
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 03810000 was valid after all)
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 03810000
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 03810000 was valid after all)
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 03810000
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 03810000 was valid after all)
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 03810000
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 03810000 was valid after all)
mimalloc: warning: mi_free: pointer might not point to a valid heap region: 03808000
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 03808000 was valid after all)
Generate::ALL (for bounding box) took 51 ms
Generate::ALL took 52 ms
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 03808700
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 03808700 was valid after all)
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 03808700
(this may still be a valid very large allocation (over 64MiB))
Generate::ALL took 62 ms
Written '.\surfaces.step'.
```

The same in release:

```
[C:\Users\xxxx\Desktop\solvespace\buildmsvc\bin\Release]solvespace-cli export-surfaces --output .\surfaces.step ..\..\..\git\test\group\translate_nd\normal_v22.slvs
```

And the output:

```
Written '.\surfaces.step'.
```

rpavlik, 2022-06-30:
yeah, it only shows up on the s390x architecture which is IBM z/Series aka system 360 successor. I have a docker environment setting up on my pc (using binfmt and qemu) now to bisect

here's the script: https://salsa.debian.org/science-team/solvespace/-/blob/master/debian/tests/surfaces

ruevs, 2022-06-30:
> I have a docker environment setting up on my pc (using binfmt and qemu) now to bisect

Call me very impressed that you know how to set up and IBM s/390 emulation with qemu and docker!

Crazy thought - if you run solvespace (the GUI) and point it to an X server on a PC would it actually work? The s/390 does not have "local" display hardware I presume?

rpavlik, 2022-06-30:
Yeah , x forwarding would probably work. I had never done it either, and am frankly surprised it works too. My first try was to install a vm, but I found instructions to do it with docker and it worked, so...

I am now basically trying to rerun the tests: on testing (bookworm), with the unstable (sid) version of the solvespace package installed (3.1 basically), doing the autopkgtests. Some weird results, like:

```
solvespace-cli export-view --output /tmp/autopkgtest.87e4t0/autopkgtest_tmp/%_view.pdf --view isometric ./test/constraint/at_midpoint/line_pt_normal.slvs
Missing (absent) translation for group-name'#references'
Missing (absent) translation for group-name'sketch-in-plane'
File ./src/resource.cpp, line 56, function LoadStringFromGzip:
Assertion failed: stream.avail_out == 0.
Message: Inflated resource larger than what trailer indicates.
Aborted (core dumped)
```

which nevertheless actually did not fail the test?

rpavlik, 2022-06-30:
btw

```sh
# does some kind of setup?
docker run --rm --privileged multiarch/qemu-user-static --reset -p yes

docker run --rm -it --platform linux/s390x s390x/debian:bookworm
```

phkahler, 2022-06-30:
I ran master on x86-64 linux, opened the test file with the GUI and exported surfaces. Got what looks like a valid .step file.

rpavlik, 2022-06-30:
Yeah, see the linked Debian bug: it's a very specific environment causing failure (architecture, system deps, and maybe cosmic rays). It works fine on more common setups and on my attempt to replicate the failed setup.

ruevs, 2022-08-05:
@rpavlik now there is a new (semi?)automatic bug:
https://www.mail-archive.com/debian-bugs-rc@lists.debian.org/msg624418.html

rpavlik, 2022-08-05:
Yeah I have to email the s390 mailing list. If they can't help I'll probably just remove the package from that architecture, it's pretty clearly a spurious problem, and I'm not sure if anybody is actually running Linux on Z series mainframes to use GUI apps like Solvespace 🤣

phkahler, 2022-08-05:
> Yeah I have to email the s390 mailing list. If they can't help I'll probably just remove the package from that architecture

I wanted to suggest that, but who am I to say what architectures Debian folks want solvespace to run on or why? It is the obvious solution, and I doubt anyone is using it for actual CAD work on that arch.

ruevs, 2022-12-04:
https://groups.google.com/g/linux.debian.bugs.dist/c/uIecAeacsAw

phkahler, 2022-12-04:
@rpavlik ??

rpavlik, 2022-12-04:
Yep, I got that email, plan to address it this week at work.

rpavlik, 2022-12-07:
OK, I am now just waiting for a sponsorship of the revised package, since I am not formally a DM or DD.

The underlying float precision or something issue on s390x remains, but at least this should unblock debian.

ruevs, 2023-09-06:
@rpavlik It seems 3.1 is in Debian now https://tracker.debian.org/pkg/solvespace - can we close this?

rpavlik, 2023-09-07:
yeah, it's not blocking debian anymore. It's not a build issue, unless there are flags to make s390x floating point act more like sse2.

closed 2025-05-16 by phkahler