solvespace/solvespace #1266: Unfollowable tutorial

andrzejtp, 2022-07-02:
### System information

- **SolveSpace version:** 3.1~70bde63c
- **Operating system:** Ubuntu 20.04

### Expected behavior

It should be possible to follow the "bracket" tutorial in a freshly installed solvespace. This is most likely the first tutorial a newcomer will attempt, so any negative experiences will discourage the potential new user.

### Actual behavior

The tutorial says:

"SolveSpace provides a quicker way to get the exact same result. Select the inside corner point of the profile, so that it is highlighted in red. Then choose Sketch → Tangent Arc at Point; a tangent arc will automatically be created at that point."

But when I select the inside corner point and choose Sketch → Tangent Arc at Point, the drawing becomes malformed and a red "zero-length edge" message appears. This is not consistent with the outcome expected in the tutorial.

### Additional information

When I chose "Sketch → Tangent Arc at Point" without any points selected and in the "Property browser" I choose "modify original entities", and then continue following the tutorial (select the point, choose Tangent Arc at Point) the outcome becomes as expected. The tutorial should be explicit about choosing "modify original entities".

phkahler, 2022-07-02:
Not sure if this is the problem, but you need to use the property browser to select the original sketch group where those lines were created before trying to select and round the corner.

rpavlik, 2022-07-06:
There should only be a single group at that point, I just followed it and it worked fine. Can you record a screencast so we can see where the instructions lead you astray?

closed 2023-09-06 by ruevs