solvespace/solvespace #1269: Solid model is missing for an assembled part that was forced to triangle mesh at design time

blinkenlight, 2022-07-10:
### System information

- **SolveSpace version:** 3.1~d6e1b230
- **Operating system:** Windows 7 x64

### Expected behavior

Triangle-mesh assemblies should have a visible solid model

### Actual behavior

Assembled part that had beed restricted to triangle mesh at design time loads with no visible solid model as an assembly

### Additional information

Using the model attached in #1268 except with ticking "restrict to triangle mesh" on the cylinder group to avoid the naked edges issue, loading it as an assembly there is no solid model of it visible. Also ticking the "restrict to triangle mesh" option in the group created by loading this model as an assembly makes the solid model appear. This is, however, extremely counter-intuitive until one finally figures out this needs to be done to see the solid model.
[image]
[image]

phkahler, 2022-07-10:
@blinkenlight I'm speculating at the pub. If you force the model to triangle mesh in the first extrusion instead of the last one, does it link properly?

blinkenlight, 2022-07-11:
Nope. No matter which group (extrusion or sketch) I use to restrict to triangle mesh, the result is the same - no solid model as an assembly unless I also restrict the assembly itself... (also: cheers! ;))

ruevs, 2022-09-14:
https://solvespace.com/forum.pl?action=viewthread&parent=4791

phkahler, 2023-01-28:
I've made a local branch to fix this. The best way to fix it seems to be to modify the function Group::IsTriangleMeshAssembly()  which just checks if we linked an STL file. A better test is if the linked group has a triangle mesh but no NURBS shell, as that should cover every case including STL and OBJ if we ever do it, while also flagging slvs files that end with a mesh but not shell.

I'm not sure how to check that the group has a mesh or shell though. Poking around the code now...

ruevs, 2023-01-29:
https://solvespace.com/forum.pl?action=viewthread&parent=5034&tt=1674951254

PR #1336 (2023-01-29): Treat group as mesh only if a linked file has a mesh and no shell.
Fixes #1269
closed 2023-01-29 by phkahler