solvespace/solvespace #1276: Export 2D View as G-Code splits arcs to multiple G01 segments instead of one G02/G03 operation

yiannikos-p, 2022-08-04:
### System information

- **SolveSpace version:** 3.1-1
- **Operating system:** Artix Linux

### Expected behavior

Trying to export 2d view as G-Code but instead of using G02/G03 the curve is split into linear segments (G01)

### Actual behavior
i.e. 
starting at 0,0 in G90 the path:
G01 X0 Y50
G03 X-50 Y0 I0 J-50
G01 X0Y0

exports as:
[thing.txt](https://github.com/solvespace/solvespace/files/9261802/thing.txt)
### Additional information
I suppose this also happens with all curves but it should at least be fixed in circles. I tried it on some stepper motors and on the first gcode file the motors slowed down and then smoothly changed direction and on the exported gcode it was noticeably less smooth.
It also makes the gcode much less readable and therefore user friendly.

Thanks in advance

Here's the slvs file so you can try it yourselves
[thing.zip](https://github.com/solvespace/solvespace/files/9261823/thing.zip)

phkahler, 2022-08-05:
I hope to get around to this in the not too distant future. From an implementation standpoint I think splitting out the gcode creation from the other 2D formats in export.cpp is a good idea and will encourage more code contributions. I'd also like to create a separate page for CAM options in the text window. I think gcode has been underappreciated in solvespace ;-)

yiannikos-p, 2022-08-05:
Thanks for answering, sorry that I opened the issue I just wasn't sure my
email had gone through. I'll take a look myself on export.cpp, but I I'm
not sure I'll manage to help a lot my oop skills are not that good.
Thanks again,
Yiannikos

phkahler, 2022-08-06:
@yiannikos It looks like the exporter is converting one or more SBL (solvespace bezier loop) to a contour which is essentially a list of points from the piecewise linear version of the curves in the SBL.

The offset function is working at the contour level, not the SBL level. The code seems overly complex and treats inside corners differently than outside even though I think they can be done the same way ;-)

To export G02 and G03 will probably require doing the cutter comp offset at the SBL level instead of the piecewise-linear level. That can work perfectly for arcs of a circle. Another issue is an offset bezier needs to be intersected with its neighboring beziers in the loop. There are already functions for supporting that kind of thing in ratpoly.cpp. For example finding a point on two curves, and reparameterizing a portion of a curve ("split at point" or something like that).

I am interested in having a function that takes 3 consecutive curves from an SBL and offsets the middle one clipping it (or extending it) to the intersections with the other two offset curves. This could be used for all curves in an SBL to offset it, and it would be useful here as well as in code for creating chamfers and fillets.