solvespace/solvespace #1278: SolveSpace version: v3 empty black window

quangduyitx, 2022-08-08:
### System information

- **SolveSpace version:**  v3.1 https://github.com/solvespace/solvespace/commit/70bde63cb32a7f049fa56cbdf924e2695fcb2916
- **Operating system:**  windows 10 64
[image]
[image]
[image]

### Expected behavior
show interface to use

### Actual behavior
an empty black window

### Additional information

ruevs, 2022-08-08:
Does one of these versions https://github.com/ruevs/solvespace/releases work? They are compiled for OpenGL 1.

If they do you should update your video card drivers and this issue would be the same as https://github.com/solvespace/solvespace/issues/1036.

quangduyitx, 2022-08-08:
> Does one of these versions https://github.com/ruevs/solvespace/releases work? They are compiled for OpenGL 1.
> 
> If they do you should update your video card drivers and this issue would be the same as #1036.

it worked, I think the error is caused by AMD Adrenalin 22.7.1 Driver
it has OpenGL API 4.6 and openGL Driver 22.0.7 Beta

ruevs, 2022-08-08:
Hmm this is the latest... https://www.amd.com/en/support/kb/release-notes/rn-rad-win-22-7-1

I'm not sure how to proceed in troubleshooting this...

phkahler, 2022-08-08:
Yikes!  That hardware is newer than my solvespace development machine - Zen1+ Raven Ridge 2400G.

quangduyitx, 2022-08-08:
> Would you be willing to downgrade to 22.5.1 and try?
> 
> https://www.amd.com/en/support/apu/amd-ryzen-processors/amd-ryzen-3-mobile-processors-radeon-graphics/amd-ryzen-3-4300u

downgrade to 22.5.1, worked.
AMD Adrenalin 22.7.1 Driver gives error
[image]

ruevs, 2022-08-08:
Very intetesting! They do mention a huge OpenGL speed up (in Minecraft) in the release notes for 22.7.1. I wonder if someone else with an AMD video card/APU will have the same problem or it is specific to this system?

jackzelenkin, 2022-08-26:
I have the same issue with rx6400 card and adrenalin 22.8.1. Downgrading adrenalin to 22.5.1 does solve it.

qutefox, 2022-09-14:
I am having the same issue. Only a back screen.
[image]
[image]
[image]

qutefox, 2022-09-15:
Downgrading to AMD Adrenalin 22.5.1 Driver worked:
[image]

SimonGolding, 2022-11-12:
Hi I'm having the same issue, so have decided to get my cross compiler (Ubuntu Mate in a VirtualBox) going to help with testing. I just finished compiling (with debug) and tested it to confirm it has the same issue as the precompiled version.

for the sake of it I load a file as was surprised to find the model showed up. The Property Browser and all tool menus are still missing but the shaded model show up and I can select surfaces.

[image]

No error messages were printed to the console. So any suggestion on where to test would be appreciated

ruevs, 2022-11-15:
This is very interesting. If you run the debug version that you built from a command prompt want messages do you see?

Another interesting test that just occured to me - if you run an [OpenGL 1 version](https://github.com/ruevs/solvespace/releases) how does it behave?

ruevs, 2022-11-15:
As for places to look at - if the OpenGL 1 version works then the offending code has to be in 

https://github.com/solvespace/solvespace/blob/master/src/render/rendergl3.cpp

Maybe here:
https://github.com/solvespace/solvespace/blob/50cbecbe7249a1a47561fca33919a08db7970e49/src/render/rendergl3.cpp#L481

and here:
https://github.com/solvespace/solvespace/blob/50cbecbe7249a1a47561fca33919a08db7970e49/src/render/rendergl3.cpp#L544

SimonGolding, 2022-11-15:
[image]

So my debug compiled version on the left and the OpenGL V1 on the right.

[image]
My system detail for completeness, I will have a look into the suggested code when i get a moment.

Asdow, 2022-11-17:
Hi. 

This same issue is plaguing several users of Blender and downgrading to the aforementioned 22.5.1 drivers fix it, so it is not restricted to only this software. The newest Adrenalin 22.11.1 release notes list "OpenGL applications using MSAA may see visual corruption." in its fixed issues, but I haven't tested to see if it affects solvespace or blender yet.

This was the issue that the driver update references.
https://www.reddit.com/r/Amd/comments/yfnknt/amd_adrenalin_edition_22103_optional_released/iu6fmfl/?context=3

K900, 2022-11-17:
It is unfortunately not fixed on 22.11.1.

ruevs, 2022-11-17:
Thanks for pointing out the problem in Blender.
Here is the issue for reference:
https://developer.blender.org/T100096
According to it 22.6.1 also works. Would one of you verify that this is the case with SolveSpace as well?

optlink, 2022-11-17:
I have the problem with 22.11.1. The web version works on Edge for me.

ruevs, 2022-11-23:
On the Blender bug someone posted a possible solution. Can any (all) of you check if it solves the problem for SolveSpace?

> [Laurent (Soy55)](https://developer.blender.org/p/Soy55/) added a comment.[Wed, Nov 23, 1:37 PM](https://developer.blender.org/T100096#1450128)
>
> Find the real culprit. you absolutely need have MSAA set as "Use appliccation settting" in the AMD driver. reverted to 22.11.1 and it work.

> [René (ShockHazard1)](https://developer.blender.org/p/ShockHazard1/) added a comment.[Wed, Nov 23, 3:22 PM](https://developer.blender.org/T100096#1450186)
>
> I can confirm the statement [@Laurent (Soy55)](https://developer.blender.org/p/Soy55/), because I tested it after his hint.
The issue is solved, if the AMD driver use for MSAA the option "application settings" or "expand application settings". This works on the fly and Blender needs just a restart.

phkahler, 2022-11-23:
This issue is affecting a lot more than Solvespace and Blender:

https://www.reddit.com/r/Amd/comments/wabgn3/opengl_msaa_wireframe_bug_in_2271_driver/

It looks to me like AMD broke something and probably needs to revert a change, and someone is yelling "but performance!"  Yeah, we all struggle to make things better without breaking stuff  ;-)

Asdow, 2022-11-23:
> On the Blender bug someone posted a possible solution. Can any (all) of you check if it solves the problem for SolveSpace?
> 

Setting Antialiasing to Use application settings fixes the bug on the newest drivers with Blender, but not with solvespace for me. I wonder if solvespace is using some sort of openGL call related MSAA by default that blender does not?

Oh, and the web version of solvespace works for me on Firefox, albeit I didn't check whether changing the antialiasing setting could reproduce the glitch there as well

ruevs, 2022-11-23:
To sum up:
The complete OpenGL rewrite in 2.7.1 breaks us. 2.11.1 and the MSAA option do not fix SolveSpace, but it fixes Blender. The Web version and the OpenGL 1 versions always work.

This leads me to believe it may be related to our rather ancient (2016) [fork of Angle](https://github.com/solvespace/angle/tree/chromium/2704+cmake) they OpenGL 3 goes through on Windows. I tried to quickly update it but it is a pain to adjust the CMake configuration... If I find the time to do it I'll post a version with "modern" Angle to test.

phkahler, 2022-11-23:
@ruevs it would be nice to use an upstream version of Angle too, and dump our old one.

majso, 2023-02-19:
fyi: I've tested with driver 22.11.2 and got same result with black screen

fred777, 2023-02-23:
Same issue with Adrenaline driver 23.2.2 on Win10 and RX6800XT. Tried different custom AA settings for solvespace without luck.

dan0h, 2023-02-23:
Same issue for me, Adrenalin driver 23.2.2, Win11, 6900XT, interestingly FreeCADLink is also borked, but if I select the "Experimental" render cache, it fixes the issue - in FreeCADLink its characterised by crazy random lines shooting from the geometry. OpenGL 1 SolveSpace works fine.

Slaw6820, 2023-02-24:
Hey guys,
as a quick workaround from this code part:
https://github.com/solvespace/solvespace/blob/50cbecbe7249a1a47561fca33919a08db7970e49/src/platform/guiwin.cpp#L635-L651

You can comment out or remove this line 
https://github.com/solvespace/solvespace/blob/50cbecbe7249a1a47561fca33919a08db7970e49/src/platform/guiwin.cpp#L641

to disable OGL ES context creation and recompile the project.
Then it works :)

ruevs, 2023-04-02:
Intel finally has released a new driver for the [AMD Radeon™ RX Vega M GH in my i7-8809G](https://ark.intel.com/content/www/us/en/ark/products/130409/intel-core-i78809g-processor-with-radeon-rx-vega-m-gh-graphics-8m-cache-up-to-4-20-ghz.html) so now I can reproduce the problem. Will post more when I have time to debug it.

Slaw6820, 2023-04-05:
@ruevs 
New AMD driver exposes new extension: WGL_EXT_create_context_es2_profile 
This causes app to use EGL_PLATFORM_ANGLE_TYPE_OPENGLES_ANGLE, but previously app was using EGL_PLATFORM_ANGLE_TYPE_OPENGL_ANGLE. Seems like app or driver has issue with the OGL ES, so per my last post, just disable it as a workaround. Then app would take the same route on both old and new driver.

> Hey guys, as a quick workaround from this code part:
>
> https://github.com/solvespace/solvespace/blob/50cbecbe7249a1a47561fca33919a08db7970e49/src/platform/guiwin.cpp#L635-L651
>
> You can comment out or [...]

ruevs, 2023-04-05:
> New AMD driver exposes new extension: WGL_EXT_create_context_es2_profile
> This causes app to use EGL_PLATFORM_ANGLE_TYPE_OPENGLES_ANGLE, but previously app was using [...]

@Slaw6820 I did try your workaround and it works. Thanks a lot!

What is bothering me is why does it fail with "EGL_PLATFORM_ANGLE_TYPE_OPENGLES_ANGLE" when on a browser the web version is probably using the same profile.

I debugged it a bit and I think I narrowed it down to a missing extension that our (ancient) Angle expects but the driver does not provide. However I am not completely sure - this is the first time I am touching the OpenGL code. I'm not on that machine right now so I can not say what the extension was.

FlashSystems, 2023-04-09:
AMD seems to have fixed the problem: After updating to Adrenalin 23.4.1 the 3.1 stable build is working again on my Radeon RX550.

Slaw6820, 2023-04-17:
@ruevs 
> I debugged it a bit and I think I narrowed it down to a missing extension that our (ancient) Angle expects but the driver does not provide. However I am not completely sure - this is the first time [...]

Curious if any progress on that. What extension is expected?

bruce965, 2023-05-01:
Hope this can be helpful to anyone still facing this issue...

A temporary solution (only for computers where the display is connected to the motherboard port and not directly to the GPU) is to disable the GPU from the Device Manager, forcing the whole system to run through CPU soft-rendering.

Another less aggressive temporary solution is to use [Mesa3D drivers for Windows](https://github.com/pal1000/mesa-dist-win), basically bypassing the GPU and using the CPU to soft-render everything. This is a per-app setting, thus it doesn't affect the whole system, and works well enough for me. This is my current approach.

Miguel-Rodrigues, 2023-05-15:
It's now working for version Adrenaline 23.4.3
[image]

[image]

ilobs, 2023-09-16:
So I seem to be having the same problem with 23.9.1 on my G14 with an CPU
AMD Ryzen 9 6900HS with an AMD Radeon RX 6700S.  Is anyone else having the same issue again?

cowboycasey, 2023-09-16:
Just found this thread and I am also having the same issue..  Thought I was loosing my mind and could not see anything...  Just updated to the newest version and still not working..  Followed the instructions above by disabling the GPU and using software and it works great..
[image]
[image]

TerryGeng, 2023-09-22:
This problem persists with 23.9.2 on AMD Radeon RX 6400.

TerryGeng, 2023-10-24:
Seems that 23.10.2 fixed this issue again.

cowboycasey, 2023-10-24:
I just loaded 23.9.2 and it is working for me without disabling..

[image]

Nakilon, 2024-08-18:
Having this on 23.7.1, RX 560.

ruevs, 2025-09-04:
For cross reference - the vcpkg version of Angle - perhaps we can use that?
https://github.com/microsoft/vcpkg/pull/46841