solvespace/solvespace #1279: SolveSpace crashes on lathe

suhr, 2022-08-09:
On Linux, SolveSpace just crashes when lathe tool is used. On Windows, it hangs first, and might crash later.

This is the terminal output on Linux:

```
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
trim was empty
trim was empty
trim was empty
trim was empty
trim was empty
trim was empty
trim was empty
trim was empty
trim was emptytrim was empty

trim was empty
trim was empty
trim was empty
trim was empty
trim was empty
File ./src/srf/surfinter.cpp, line 103, function AddExactIntersectionCurve:
Assertion failed: !(sb->Start()).Equals(sb->Finish()).
Message: Unexpected zero-length edge.
trim was empty
```

## How to reproduce

Open [keycap.slvs.zip](https://github.com/solvespace/solvespace/files/9292796/keycap.slvs.zip), select a normal and a point and invoke `Lathe`.

### System information

**SolveSpace version:** 3.1
**Operating system:** both Windows and NixOS

### Additional information

The sketch contains an arc of a quite large circle. But it's required to make a rounded keycap.

phkahler, 2022-08-10:
First I've attached a somewhat fixed version of the file:
[keycap.zip](https://github.com/solvespace/solvespace/files/9302992/keycap.zip)
.
I made 2 changes. One is I raised the horizontal line in the 2D sketch and added a new vertical line near the center. This prevents the lathe group from having 2 surfaces tangent to the hex flat surface - a classic tanget-surfaces issue with solvespace. But I also had to leave a flat spot in the middle (which you don't want). You can drag that point radially in the attached sketch to see that at small radius we get a NURBS failure. I'm not sure what's causing that - it could be related to the first failure or something different.

A better work around may be to add height to the hex shape and let the lathe cut the entire top off, so there is no tangency between the lathe-arc and the flat surface.

@ruevs  so one duplicate NURBS issue, but also maybe a new one? We should be able to make an even simpler test case for this, but this is a relatively small sketch.

suhr, 2022-08-10:
I sort of workarouned this by forcing NURBS to use triangles and decreasing chord tolerance.

phkahler, 2022-08-10:
It's not #1058 because that is due to overlapping surfaces caused by doing lathe on a full circle to create a sphere. This sketch only has entities on the right side of the lathe axis. My first thought was the point on the axis producing NURBS patches with a degenerate edge, but it still happens if we move that point off-axis.

Thought: what would happen if we just skipped adding the zero-length exact intersection curve rather than crashing? Also not sure that test is as robust as it looks (control points are not checked).

ruevs, 2022-08-27:
I think it is still the same problem since the sphere is tangential to the plane and the thickness approaches zero toward the center.

ruevs, 2023-04-07:
[On the forum](https://solvespace.com/forum.pl?action=viewthread&parent=5211) a user posted one more reproduction case.

To reproduce open "volumechambredecombustion4.slvs" from the attached archive and lathe "g005-sketch-in-plane" in such a way as to produce the shape in "volumechambredecombustion.slvs".
[AnotherLatheZeroLengthEdge.zip](https://github.com/solvespace/solvespace/files/11178069/AnotherLatheZeroLengthEdge.zip)

[image]

The call stack is as usual:
```
solvespace.exe!SolveSpace::Platform::FatalError(const std::string & message) Line 182
solvespace.exe!SolveSpace::AssertFailure(const char * file, unsigned int line, const char * function, const char * condition, const char * message) Line 15
solvespace.exe!SolveSpace::SSurface::AddExactIntersectionCurve(SolveSpace::SBezier * sb, SolveSpace::SSurface * srfB, SolveSpace::SShell * agnstA, SolveSpace::SShell * agnstB, SolveSpace::SShell * into) Line 103
solvespace.exe!SolveSpace::SSurface::IntersectAgainst(SolveSpace::SSurface * b, SolveSpace::SShell * agnstA, SolveSpace::SShell * agnstB, SolveSpace::SShell * into) Line 344
solvespace.exe!SolveSpace::SShell::MakeIntersectionCurvesAgainst(SolveSpace::SShell * agnst, SolveSpace::SShell * into) Line 718
solvespace.exe!SolveSpace::SShell::MakeFromBoolean(SolveSpace::SShell * a, SolveSpace::SShell * b, SolveSpace::SSurface::CombineAs type) Line 808
solvespace.exe!SolveSpace::SShell::MakeFromUnionOf(SolveSpace::SShell * a, SolveSpace::SShell * b) Line 13
solvespace.exe!SolveSpace::Group::GenerateForBoolean<SolveSpace::SShell>(SolveSpace::SShell * prevs, SolveSpace::SShell * thiss, SolveSpace::SShell * outs, SolveSpace::Group::CombineAs how) Line 194
solvespace.exe!SolveSpace::Group::GenerateShellAndMesh() Line 407
solvespace.exe!SolveSpace::SolveSpaceUI::GenerateAll(SolveSpace::SolveSpaceUI::Generate type, bool andFindFree, bool genForBBox) Line 277
```