solvespace/solvespace #1281: Collection of NURBS and Mesh Problems

BKLronin, 2022-08-19:
### System information

- **SolveSpace version:** 3.1 70bde63c
- **Operating system:** Manjaro Ruah 21.36

Just as mentioned in the forums. A small collection of models were I had to force to mesh and partly still had some problems. 

Mostly on Extrude Cut operations after several other features. I already tried to get around known problems.
[Archiv.zip](https://github.com/solvespace/solvespace/files/9383224/Archiv.zip)

Multiple cutouts meeting features.
[ShovelStiel_Ultralighr.slvs.zip](https://github.com/solvespace/solvespace/files/9383232/ShovelStiel_Ultralighr.slvs.zip)

More cut operations, examples of working and problematic.
[Archiv_morecutouts.zip](https://github.com/solvespace/solvespace/files/9383259/Archiv_morecutouts.zip)

ruevs, 2023-09-22:
Very interesting models! And I really like how you used `suppress this group's solid model` (e.g. in BV6000_halter.slvs in [Archiv.zip](https://github.com/solvespace/solvespace/files/9383224/Archiv.zip)) to leave extrusions that cause NURBS errors around after finding a workaround.

BKLronin, 2023-09-22:
Glad they are of use. :)

I´ll try to remember to keep these in for comparison in future models if it helps.

ruevs, 2026-11-03:
## Status with #1806

[Archiv.zip](https://github.com/solvespace/solvespace/files/9383224/Archiv.zip):
- Action.slvs - fixed in 3.3
- BV6000_halter.slvs - fixed... mosly in 3.3
- Gepaeckhalter front_kurz.slvs - fixed in #1806
- TrinkflascheQuechua1L.slvs - fixed in 3.3

[ShovelStiel_Ultralighr.slvs.zip](https://github.com/solvespace/solvespace/files/9383232/ShovelStiel_Ultralighr.slvs.zip)
- ShovelStiel_Ultralighr.slvs - fixed in #1806

[Archiv_morecutouts.zip](https://github.com/solvespace/solvespace/files/9383259/Archiv_morecutouts.zip)
- bEHRINGERDOUBLE.slvs - fails with `g00e` enabled
- Keyboardunterbau.slvs - fixed in 3.3
- MicroFreak.slvs - fixed in 3.3
- Neutron_halter.slvs - fails with triangle mesh off in `g00a`