solvespace/solvespace #1282: Quick dragging of sketch could broke it

mkyral, 2022-08-20:
### System information

- **SolveSpace version:** 3.1~c65e31be
- **Operating system:** Gentoo Linux

### Expected behavior

I've noticed, that sometimes, when I dragging sketch around workplane, sketch could break. It looks like it always happen, when I drag sketch too quickly.

### Actual behavior
1) create simple test case - rectangle with construction line inside
[image]

2) click to some line and slowly drag model around workplane - all is ok
3) do it again, but this time drag it quickly down - construction line could move outside rectangle
[image]

4) Point is still constrained to the line
[image]

The same issue is with arches
[image]
[image]

### Additional information

Both sketches attached in file: [slvs_drag_issue.zip](https://github.com/solvespace/solvespace/files/9387729/slvs_drag_issue.zip)

ruevs, 2022-08-20:
Known. https://github.com/solvespace/solvespace/issues/76

As workaround:
 Temporarily `relax constraints and dimensions` from the group properties in the text window, "straighten" the sketch roughly and then turn the option off.

Or just undo ;-)

closed 2022-08-20 by ruevs

phkahler, 2022-08-20:
Or for some cases just yank it back.