solvespace/solvespace #1283: Exporting to .STEP creates a file that isn't recognised by several online services

garethbowker, 2022-08-21:
### System information

- **SolveSpace version:** 3.1~70bde63c
- **Operating system:** macOS Ventura Beta

### Expected behavior

Exporting a file to STEP format creates a file that can be opened by Fractory.com / Xometry.eu / STEPViewer3D (Mac App https://apps.apple.com/gb/app/step-viewer-3d/id776843646?mt=12 )

### Actual behavior

A file is created, but it is not recognised as a STEP file by several web services and a STEP viewer app.

### Additional information

[Fuel Panel.zip](https://github.com/solvespace/solvespace/files/9388747/Fuel.Panel.zip)

ruevs, 2022-08-21:
A possible problem could be that the STEP files that SolveSpace exports are not "solid". See this issue https://github.com/solvespace/solvespace/issues/206 and try the 10mm cube [and the modified one](https://github.com/solvespace/solvespace/issues/206#issuecomment-296350770) and see if it makes a difference.

garethbowker, 2022-08-21:
Thanks @ruevs, this looks to be a duplicate of #206. The 10mm cube opens correctly in STEPViewer3D. I'll give @nanoant's branch a go and see if that manages to export in a file format I can then import.

closed 2022-08-21 by garethbowker