solvespace/solvespace #1284: Tangent without coincicence?

shooshx, 2022-08-21:
Reading the library doc:
https://github.com/solvespace/solvespace/blob/master/exposed/DOC.txt

It occurred to me that all the tangent constraints
SLVS_C_ARC_LINE_TANGENT
SLVS_C_CUBIC_LINE_TANGENT
SLVS_C_CURVE_CURVE_TANGENT
contain some implicit point coincidence constraints so that the tangent is at the end or begining of a segment
This make sense if what you're building is a closed polygon, but that's not true in my case.
If you look at what Fusion 360 does for instance, you can have tangent curves that don't have a constraint on a shared point. Just an arc touching a line anywhere on the line for instance. This gives a lot more flexibility and you can always add the missing point coincidence constraints where they are needed.

Is it possible to get the same tangent constraints as above but without the point coincidences?

ruevs, 2022-08-21:
That used to be the case. Was confusing for someone and got "fixed"/changed...

ruevs, 2022-08-21:
OK I did not remember the situation completely correctly. Here is the issue https://github.com/solvespace/solvespace/issues/937

closed 2022-08-21 by ruevs