solvespace/solvespace #1287: feature request: Math curves

nick87720z, 2022-09-02:
For now curves can be only NURBS or arcs (with arcs limited to workplane). Besides that - curves could be built by function. There could be choice between coordinate systems (e.g. polar to create complex enclosed shape), though that's not main example I mean. In 3d, for example, in could be used to generate any arbitrary kind of helix (spring, spiral, etc).
It would be most useful with already planned loft group tool.
If you are interested about real use - I wanted to model helix section, which starts expanding in diameter, but ends narrowing.

Evil-Spirit, 2022-09-29:
Please check NoteCAD. There is arbitrary curves you can create - simple like sin / cos and hardcore like catenary arch or even Freshnel integral based clotoid. Say hello to real evolvent profiles!

nick87720z, 2022-09-30:
> Please check NoteCAD.

Thanks, perhaps will look. I achieved result by concatenating two springs with different diameters, with lengths controled by axial 3d lines with ratio limitation between them. First modeling them in crossectional 2d sketch, with some axial lines bound to some key points to bind resulting helix to this sketch. It looks horrible only for first time - as they say, eyes in scare, hands in action.

Though I'm actually trying again freecad due to problems with crossing (mainly) curve surfaces, like when you intersect or subtract to cylinders with perpendicular axises. Forcing NURBS to faces just doesn't look fine, yet multiplies memory consumption (last time I tried to turn on nurbs to faces feature, it froze for long due to swap).

Evil-Spirit, 2022-10-07:
> Forcing NURBS to faces just doesn't look fine

It is strongly recommended to move SolveSpace to use OpenCASCADE, so this will be less buggy.

nick87720z, 2022-11-21:
Just to note, OpenCASCADE is also not ideal. I tried loft between triangle and circle, than make a shell with one end opened. One of two ends (or both for certain) caused some mesh errors if selected to be opened. These errors could be avoided by decreasing shell size, but than I got intersections between inner and outer meshes.

phkahler, 2022-11-28:
Solvespace only supports one type of curve - rational b-splines up to degree 3. Surfaces are NURBS up to degree 3x3. Anything else would require a LOT of internal work including areas that are still buggy like booleans.

closed 2022-11-28 by phkahler