solvespace/solvespace #1288: Duplicate operations - lathe and revolve

nick87720z, 2022-09-03:
Just one revolve operation is enough.
There could be switch for angle mode with example values "Lathe" and "Partial". Of course, current way of getting lathe-like result by overrotating revolution works, but it leaves redundant control points, which contribute to clutter, so in my opinion - mode switch should have better usability.

ruevs, 2022-09-03:
Lathe existed back to version 1.9 at least. Revolve was added later. the reason both exist is backward compatibility.

closed 2022-09-03 by ruevs

nick87720z, 2022-09-03:
Btw, I just realized - no need to change file format. Change to UI would be enough, so no breakages are possible even theoretically.

Yet must admit - I made this proposal only because localized action names for these two are same for me (though icons are fine - hard to mix up).

ruevs, 2022-09-03:
> localized action names for these two are same for me (though icons are fine - hard to mix up).

Which language?

nick87720z, 2022-09-03:
Russian