solvespace/solvespace #1289: required submodules

albfan, 2022-09-04:
### System information

- **SolveSpace version:** 3.1
- **Operating system:** Fedora 36

### Expected behavior

build from source

### Actual behavior

missing eigen/spacecore include

### Additional information

README instructions try to describe minimal submodules to clone:

    git submodule update --init extlib/libdxfrw extlib/mimalloc

although seing submodules probably most people with do a simple:

    git submodule update --init

or

    git clone --recurse-submodules

looks now eigen is a minimal requirement:

git submodule update --init extlib/libdxfrw extlib/mimalloc extlib/eigen

is that worth for a PR on README.md?

ruevs, 2022-09-04:
@albfan where are you looking at the readme from? Because just last week I updated it to split out the checkout instructions https://github.com/solvespace/solvespace/commit/bce25bb0e2811d5a1b40d305d0e35ec7252d58b1. But even before that the instructions already included eigen or just `init`.

albfan, 2022-09-05:
I check last stable version 3.1

So fixed then

closed 2022-09-05 by albfan