solvespace/solvespace #1291: Rounded corner cut (fillet) issue

phkahler, 2022-09-06:
### System information

- **SolveSpace version:** All
- **Operating system:** All

### Expected behavior

It should be possible to create a fillet by using a difference operation as shown:
[cube_cut.zip](https://github.com/solvespace/solvespace/files/9498557/cube_cut.zip)

[image]

### Actual behavior

This file has curve control points P1 and P2 in group g004-sketch-fillet as well as the "corner" point of the fillet. When the corner point and P1 are coincident with the top cube edge (use point on line constraints) the NURBS fail.  Having just one of them on the cube edge is not enough to cause the problem.

The simplest work around for creating a fillet by cutting an arc is to place the corner point further away from the cube.

### Additional information
I made this sketch for easy experimentation with this issue. The curve-face tangency appears in several NURBS related issues, so this may be redundant but still a simple test case.

Since the tangent curve works with the corner point off-cube, and the corner point on-cube works if the curve is not tangent, this does not seem like a numerical issue (like finding the surface intersections) but more like a logic issue in determining which curves and surfaces to keep during the boolean operation. It is also notable that there is no failure if we switch from difference to an intersection boolean, even if the points are all colinear. Union also fails. It might also be a bug in merging the coincident surfaces. So somewhere in here:
https://github.com/solvespace/solvespace/blob/master/src/srf/boolean.cpp
https://github.com/solvespace/solvespace/blob/master/src/srf/merge.cpp

ruevs, 2022-09-06:
Good example! I'll add it to the tracking issue whem I am on a computer (next week). In addition I have an example involving only straight lines that passes with intersection and fails with difference and union. I created it two years ago when I implemented intersection, by have not gone back to it.

There seem to be at least five classes of "NURBS" failures that are emerging and there are more than enough examples in each group.

Off the top of my head (read edges as curves based on taste/case)
1. Zero length edges
2. Incorrectly split edges
3. Other bad intersections (numerical?)
4. Incorrectly selected edges
5. Incorrectly selected "patches" to keep.

I've only really ever looked at the code related to 1, 4, 5 and it always takes me a lot of time and thinking to even get into it... So I rarely start :-(

phkahler, 2022-09-07:
@ruevs,  Another option with this file is to make the final extrude two-sided to change the topology a little. You had gone further into type 4 and 5 issues than me, but I am getting interested in this particular one for some reason. I still don't understand keepEdge or whatever it is that determines what to keep and discard. IIRC every edge that is a result of surface-surface intersection (and is inside the polygon on both) needs to be kept. In this case, the ends of the curved surface should always be found as "intersection" with the cube faces via the new "trim curve is entirely in plane" case I coded a while ago [here](https://github.com/solvespace/solvespace/blob/master/src/srf/surfinter.cpp#L316)

ruevs, 2022-10-18:
I debugged this a bit. I think an edge is not created during the splitting. There should be 4 edges where the cylinder meets the face of the cube tangentially but there are only three. I need more time...

ruevs, 2022-10-23:
Debug code https://github.com/ruevs/solvespace/tree/NURBS_debugging

And test models:
[CylinderTangentFromTheInside_AKA_Fillet_NURBSTests.zip](https://github.com/solvespace/solvespace/files/9846083/CylinderTangentFromTheInside_AKA_Fillet_NURBSTests.zip)

phkahler, 2022-10-25:
@ruevs  I wonder the the missing edge is not found, or if it is discarded, or something else.

ruevs, 2022-10-25:
> I wonder the the missing edge is not found, or if it is discarded, or something else.

It ends up "unclassified" here https://github.com/solvespace/solvespace/blob/3dc4d0e640666b27abe5123979f89717bd0ae06a/src/srf/raycast.cpp#L509
But I need to think what is the right thing to do in general. The dot products end up

`dotp[0] > DOTP_TOL && dotp[1] < -DOTP_TOL`

and this is a missing case.

phkahler, 2023-03-24:
> It ends up "unclassified" here
>
> https://github.com/solvespace/solvespace/blob/3dc4d0e640666b27abe5123979f89717bd0ae06a/src/srf/raycast.cpp#L509
>
> But I need to think what is the right thing to do in [...]

@ruevs  are you sure thats where it fails? I would think it gets handled in this case:
https://github.com/solvespace/solvespace/blob/3dc4d0e640666b27abe5123979f89717bd0ae06a/src/srf/raycast.cpp#L479

The edge is splitting an existing face. If this is an edge joining the fillet to the flat face it's going to be classified as coincident with both surfaces. Does that mean coincident with both shells? It's not quite clear to me what represents what here. But I think one of those should end up classified as INSIDE even though based on edge and normals they look like coincident surfaces.

We need better documentation on what surfaces the edge joins and which shell. An intersection edge will join one surface from each shell. I can't tell what's what in that code. We may need more infomation about curved surfaces to get it right.

ruevs, 2023-03-24:
It's been a while but I wrote **`dotp[0] > DOTP_TOL`** in my comment and the if statement you cited is `fabs(dotp[0]) < DOTP_TOL` so it does not seem to cover it. I need to come back to this.

The meaning of "indir"/"outdir", shell/original, etc., etc. in that code is certainly not obvious and it always takes me some time to get up to speed. What helps me is the rather... baroque... NURBS debugging code I've accumulated over time - see the link a few comments above. The latest addition is `DumpEdgeClassificationVennDiagram`.

phkahler, 2023-03-25:
I think ultimately we are going to need to evaluate the curvature of surfaces - in a given direction at a point.  This code seems to be deciding if the surface from the edge is inside/outside, or coincident with another surface. Everything we've got at a point on the edge is going to indicate the 3 surfaces are coincident, when we really want to know that one of them bends into the other shell.

I grabbed my copy of the NURBS book and "curvature" is not in the index. Maybe I still have my old calc book which I believe covered curvature at least for parametric curves.

phkahler, 2023-04-01:
> I debugged this a bit. I think an edge is not created during the splitting. There should be 4 edges where the cylinder meets the face of the cube tangentially but there are only three. I need more time...

I've been looking at the case shown below. There should be 3 edges at the sharp edge of the fillet. One as part of the fillet shell itself, and one joining each of the 2 fillet surfaces to the flat side of the box.
[image]

Notice that the curved surface is present! I think that means the edge was kept. Does that mean it's discarding the region of the cube face that we see through and nothing else? This is also a case where intersection works but difference and union fail. It also works fine if the curve is not tangent - move the control point off the cube edge/face.

phkahler, 2023-04-05:
I put a bunch of debug prints in SSurface::MakeCopyTrimAgainst()  Looking for where it discards an edge belonging to a higher order surface (the above contains one where degm+degn == 4). I was not able to find ANY such edges discarded (be careful these are new intersection edges but either way...)

So now I'm thinking the 2 new intersection edges are being discarded by this:
```
    // Cull extraneous edges; duplicates or anti-parallel pairs. In particular,
    // we can get duplicate edges if our surface intersects the other shell
    // at an edge, so that both surfaces intersect coincident (and both
    // generate an intersection edge).
    final.CullExtraneousEdges(/*both=*/true);
```
This would mean it's not really an edge being discarded wrongly by the fancy tests, but NOT being discarded by the fancy tests. Then both intersection edges would be thrown out here. But this still doesn't make sense, as there should be 3 edges - two intersections and one from the fillet shell. I just wanted to document what I saw, or did not see so far.

drwho495, 2025-08-28:
hi there, i did some debugging on this issue and found a few things, the edges that are not being included are in the interception loop in `MakeCopyTrimAgainst` and the reason for it not being included is because the `indir_shell` variable is set to "SURF_COINC_SAME" instead of "SURF_OUTSIDE" (so i assume a change needs to be made by `ClassifyEdge` to add this edgecase)

[FilletTangentTests.zip](https://github.com/user-attachments/files/22015251/FilletTangentTests.zip)

is there any documentation on the nurbs kernal that i can look at to figure out what im reading?

ruevs, 2025-08-28:
@drwho495 indeed it is ClassifyEdge that is failing see the very fresh discussion (literally from the last few days) on this topic starting with this https://github.com/solvespace/solvespace/issues/1619#issuecomment-3225306245

drwho495, 2025-08-28:
> @drwho495 indeed it is ClassifyEdge that is failing see the very fresh discussion (literally from the last few days) on this topic starting with this https://github.com/solvespace/solvespace/issues/1619#issuecomment-3225306245
> 
> As for documentation - no formal documentation exists but if you look at issues an pull requests labeled "documentation" and "NURBS" you will find many useful comments.

Ok, thanks