solvespace/solvespace #1297: Re-linking a missing linked file only lets you select .slvs files, so it doesn't let you set the new path for a linked .stl file

interfect, 2022-10-05:
You can create a link to a `.stl` file in an assembly, but if the link breaks because the original file moves, you can only replace it with a `.slvs` file.

The problem is that the re-linking dialog uses `SolveSpaceModelFileFilters`:
https://github.com/solvespace/solvespace/blob/bce25bb0e2811d5a1b40d305d0e35ec7252d58b1/src/file.cpp#L921
Which only lets you pick `.slvs` files:
https://github.com/solvespace/solvespace/blob/f6bb0a2d35ba6c67a72170413207d0f28bac216d/src/platform/gui.cpp#L84-L86
But when making a *new* link, you get to use `SolveSpaceLinkFileFilters`:
https://github.com/solvespace/solvespace/blob/026f9369894bd526e049ee0fbf2ee928ffdde612/src/group.cpp#L306
And that lets you pick other types, like `.stl`:
https://github.com/solvespace/solvespace/blob/f6bb0a2d35ba6c67a72170413207d0f28bac216d/src/platform/gui.cpp#L88-L93

I *think* that just changing the filters used in the open dialog might solve the problem, but I'm not sure if anything special happens when initially linking a non-Solvespace file that would need to re-run on re-linking but isn't set up to do so.

interfect, 2022-10-05:
@phkahler You seem like you might know what the right answer is here.

closed 2022-10-21 by ruevs

ruevs, 2022-10-21:
I merged the pull request fixing this - it is the correct behaviour. But even before that (at least on Windows) it was possible to "re-link" moved or renamed STL and EMN files by simply typing `*.*` in the file name field.

commit 3dc4d0e6 (2022-10-21): Allow all linkable files as missing files to fix #1297
PR #1304 (2022-10-21): Allow all linkable files as missing files to fix #1297
This appears to fix #1297. Using this on top of #1296, I was able to fix up my `.slvs` file with desktop portal paths to STLs to use proper relative paths instead.

I still see extra dialog boxes about how the original, desktop portal path files could not be read, but the re-linking works and my file seems fine afterward.