solvespace/solvespace #1303: development documentation

ZehuaZhang, 2022-10-17:
Is there any documentation around development, is there some documentation around code structure, logic, and flow, and build system, git scripts and workflow other than just looking at the code for new comers who want to contribute?

ruevs, 2022-10-17:
See here https://github.com/solvespace/solvespace/issues/1222

closed 2022-10-17 by ruevs

ruevs, 2022-10-17:
> A side question: I thought gtkmm is cross platform, but there's gtkmm code for each platform which is platform dependent?

The platform specific code is here:

[guigtk.cpp](https://github.com/solvespace/solvespace/blob/master/src/platform/guigtk.cpp)
[guihtml.cpp](https://github.com/solvespace/solvespace/blob/master/src/platform/guihtml.cpp)
[guimac.mm](https://github.com/solvespace/solvespace/blob/master/src/platform/guimac.mm)
[guinone.cpp](https://github.com/solvespace/solvespace/blob/master/src/platform/guinone.cpp)
[guiwin.cpp](https://github.com/solvespace/solvespace/blob/master/src/platform/guiwin.cpp)

Each platform uses it's own "native" GUI toolkit.

ZehuaZhang, 2022-10-17:
So the gtkmm is platform dependent, I thought it's c++ framework works for across the platforms

phkahler, 2022-10-18:
@ZehuaZhang  GTK can be used on Windows and osX but we only use it on Linux (and BSD I guess). Solvespace uses platform specific GUI toolkits.

I would like to move the GTK version to GTK4 for Linux, but we could add that as an option and keep GTK3 too.

ruevs, 2022-10-18:
@ZehuaZhang GTK is cross platform, but is a "heavy burden" - literally in terms of megabytes - on Windows. The GTK runtime that we will have to distribute is bigger than SolveSpace itself.

What is more SolveSpace was written for Windows only in the beginning so it has had the native Win32 API from day one. Linux (and later macOS) support was added in 2013. If you are interested in past GUI toolkit discussions you can read them in the issues linked from the "Alternative UI toolkit/framework?" section of the pinned issue #1024.

ZehuaZhang, 2022-10-18:
@phkahler @ruevs Makes sense. For the GUI, I actually saw the [discussion](https://github.com/solvespace/solvespace/issues/263) around `imgui`, I know it would add not only `imgui` but also a separate surface/window library, e.g. `glfw`. That would also be quite an investment initially for implementation, maybe something to look at in the future, for platform dependent gui implementations are still able to scale for current solvespace I guess