solvespace/solvespace #1305: Surface experiments

BKLronin, 2022-10-27:
3.1 70bde63c Version on Manjaro.

While drawing stuff my old friend surface problem showed up. This time I documented it a little for any debug use. 

Subtractive chamfer on corner drawn from the inside (0,0) worked but did not work from outside (inf,inf).  

[image]
[image]
When drawing the other side also from the inside what previously worked: 

[image]

But partial is good:

[image]

This is inconsistent as the with a little shaking of the point or going in or out of extrusion the red warning disappears. This fuzzy state goes away when the chord tolerance is raised to 0.01 from 0.1 and 10 segs instead 20.
Strange edges it got there too:

[image]

The first example lets me think of a failed inside outside detection for surface creation?

ruevs, 2022-10-27:
Hey I remember your bggy 3D printer :-)

This failure case looks interesting. Would you be willing to create a simple reproduction model and attach it? This way I will be able to at least figure out if it is in one of the known categories.

BKLronin, 2022-10-27:
Haha yeah that was a nice test printer for SolveSpace. Might make my final design in it one day. :)

Made a minimal repro if I understood that correctly. Iǘe attached that minimal version and the full model where the error stretchd to the larger open section on the side. I got the red warning along the edge only before I constrained the difference extrusion to the inner face :

[image]

Afterwards no problems, thats different to the big assembly. Maybe there is some pattern recognizable. I deleted and reversed a few things in the bigger model but reproduced steps as closely as possible in the repro.

[Cluster_set.zip](https://github.com/solvespace/solvespace/files/9883625/Cluster_set.zip)

Attached are three models:
- Version were the error occured
- Isolated problem area, reproduction
- Solved model, final.

Hope this helps!

ruevs, 2022-10-31:
Here is minimal reproduction case.
[1305MinimalPrizmCamferNURBSBug.zip](https://github.com/solvespace/solvespace/files/9902223/1305MinimalPrizmCamferNURBSBug.zip)
It is a known problem, but I have to find the correct [existing issue](https://github.com/solvespace/solvespace/issues/738).

ruevs, 2022-10-31:
In the meantime take a look at #1291 and https://github.com/solvespace/solvespace/issues/1298#issuecomment-1272340182 they are related, but not the same. The solution is the same though - make the triangles cutting the chamfer stick out a bit.

BKLronin, 2022-10-31:
[image]
[image]

Two cases on one edge and as soon as its only one direction it works. Probably well known and handled. Just in case.

ruevs, 2022-11-01:
Here are previous models with similar failure cases:
https://github.com/solvespace/solvespace/issues/35#issuecomment-531602706
https://github.com/solvespace/solvespace/issues/537#issuecomment-669068154
https://github.com/solvespace/solvespace/pull/594#issuecomment-625587508
https://github.com/solvespace/solvespace/issues/1091#issuecomment-903661168

BKLronin, 2022-11-01:
Thanks,
ill dig through, interesting reads. 

Optional question:
I used intersection a few times on bodys were its applicable. It has fewer conflicting edges as its essentially creating a complete new set of shells and doesn´t have to deal with exiting edges so much? So the main challenge is to find a way to find out if an edge is inside or outside while they might directly are in the same spot coordinate wise? Thats why you cant cast a ray as it is send from zero towards zero and both elements are essentially 0,0,0 to each other so there is no decision if its inside or outside?

I spend a lot of time with ray casts and path finding in 2D game development but thats about it. 

https://raise-uav.com/reakt/reakt.html (unfinished poc)

ruevs, 2026-11-03:
All fixed in 3.3.

closed 2026-11-03 by ruevs