solvespace/solvespace #1312: Some thoughts after 100+ hours usage

nowakf, 2022-12-01:
First, it's worthwhile to mention that  the reason why I've used this program so much (certainly more than 100 hours) is because it's very good. 

That said, I wanted to compile a list of the most common problems I have, as a user, and further, to sound out how you guys feel about me going and writing some patches to fix them.

1.  Invalid NURBS surfaces. My intuition is that most of the invalid surfaces arise when you have two solids, and the edge of one solid rests exactly on the edge of another solid. The fix here is usually to offset the constraint so the solids slightly extend into eachother, which is a pain.

Obviously, I'd like to be running on more than intuition here, so I'd like to try and characterize and collate what geometry leads to invalid NURBs, 

2. It's often hard to see the geometry you're interested in. Allowing users to hide selected objects would be really nice here.
3. Variadic selections (I'm aware this has been mentioned before) would be a huge timesaver.

Any thoughts, ideas, stuff I've overlooked?

phkahler, 2022-12-01:
>     1. Invalid NURBS surfaces. My intuition is that most of the invalid surfaces arise when you have two solids, and the edge of one solid rests exactly on the edge of another solid. The fix here is usually to offset the constraint so the solids slightly extend into eachother, which is a pain.

@nowakf  The master NURBS issue tracker is #738 
Having said that, you are correct about most of the problems involving coincident geometry (see #540 for an example that is now "fixed") but more specifically when surfaces are tangent at the intersection as in #1291. In that issues, @ruevs has highlighted a specific line of code that seems to be "wrong", but it's not clear what the "right" solution is. The edge should be found by the #540 fix, but seems to be discarded in the linked code when the 3 surfaces are tangent at that edge.

There are some issues like #1091 that fail depending on the length or depth of the geometry rather than the topology. In my mind I call these "proximity" issues, where only distances seem to affect weather it fails. Also in that issue is an example here:  https://github.com/solvespace/solvespace/issues/1091#issuecomment-901393520  that I looked into a bit and it seems to involve failure to split trim curves properly. My test "fix" kind of verified that but wasn't a real solution.

You'll need more understanding of the general approach to boolean operations in solvespace to make progress on this. We definitely welcome anyone to try to improve things here, so please ask more specific questions in any of the open issues where you'd like to understand better.

>     2. It's often hard to see the geometry you're interested in. Allowing users to hide selected objects would be really nice here.

Have you tried "Suppress this groups solid model" in the property browser? When you use this, the group name should appear greyed out in the main list of groups so its easy to find again.

>     3. Variadic selections (I'm aware this has been mentioned before) would be a huge timesaver.

This is IMHO a high priority usability issue. There are 2 open issues on it and 2 or 3 PRs but we need to get the implementation right, for some definition of right. Look around at the PRs and the issues in the 4.0 milestone for some discussion.

> Any thoughts, ideas, stuff I've overlooked?

TBH I thought your post was going to be a LOT longer, since there are many ways solvespace could be improved.

ruevs, 2022-12-01:
@nowakf on NURBS - @phkahler already gave you a very good overview. Here is one more "simple" and interesting case https://github.com/solvespace/solvespace/issues/1312#issuecomment-903661168 and a very unscientific classification of failures https://github.com/solvespace/solvespace/issues/1291#issuecomment-1238608356 from me.

> Any thoughts, ideas, stuff I've overlooked?

The [pinned issue](https://github.com/solvespace/solvespace/issues/1024) has a general list of commonly requested things/bugs/features that so far no one has had time/desire/skill/inclination to implement ;-)

ruevs, 2023-01-06:
By the way there is an issue on #2 already https://github.com/solvespace/solvespace/issues/1007.
And "Variadic" constraints were just merged (#1314) so I will close this.

closed 2023-01-06 by ruevs

ruevs, 2023-02-12:
#1328 0 just for cross reference ;-)