solvespace/solvespace #1319: Cannot open dialog windows when installing via snap

guija, 2023-01-04:
### System information

- **SolveSpace version:** solvespace 3.1~70bde63c
- **Operating system:** Ubuntu 22.04

### Expected behavior

When e.g. trying to save a file the "Save file" dialog should appear.

### Actual behavior

Nothing happens and Solvespace freezes.

### Additional information

This is the output I get when starting Solvespace from the command line:
`(solvespace:42239): GLib-GIO-WARNING **: 02:06:57.388: Error creating IO channel for /proc/self/mountinfo: Permission denied (g-file-error-quark, 2)
`
I found a similar issue for Inkspace on Github. Apparently it is a problem with snap package configuration?
See https://gitlab.com/inkscape/inkscape/-/issues/2495 and https://gitlab.com/inkscape/inkscape/-/issues/2540

ruevs, 2023-01-06:
I'm not on Linux, but I think this is the problem?
https://github.com/solvespace/solvespace/pull/1296
Perhaps @interfect can answer?

guija, 2023-01-07:
Yeah true, I think its a duplicate, I missed that other issue. I also added information how I installed solvespace (Using snap, see above).

ruevs, 2023-01-07:
Did the workaround suggested in the other issue help?

ruevs, 2023-01-07:
Does the workaround suggested in the other issue help?

interfect, 2023-01-08:
70bde63c is from May 2022, so it doesn't have anything I did, and #1296 wasn't even ever merged, so nothing I did *caused* this.

But it seems weird for this to be #1067 again, because that's caused by the desktop portal just not being there on the host system, and Ubuntu 22.04 would almost certainly ship `xdg-desktop-portal-gtk` in the default system. (Missing that would also break the save dialog in e.g. Firefox, I think.)

guija, 2023-01-08:
I just checked, `xdg-desktop-portal-gtk` is already installed, then I checked whether it is running with `systemctl --user status xdg-desktop-portal-gtk.service` like mentioned in https://github.com/solvespace/solvespace/issues/1067 which outputted that it is running. Trying afterwards however the dialogs _worked_. Really strange behavior. I also restarted my machine to double check and everything is still working. Really interesting.

ruevs, 2023-01-11:
No need to open a new issue https://github.com/solvespace/solvespace/pull/1323.