solvespace/solvespace #1320: message box apocalypse on split lines when endpoint is on line

Dromantor, 2023-01-05:
### Preamble

Yes, I know, I'm using the "Split lines / ..." function not in the correct way; I'm really missing the trim-line-to-next-point function most CAD programs have *; but this is far from the failure mode I expected .... it didn't do that previously and just created a zero-length edge instead of crashing after a message box invasion.

\* some things are much easier and faster to draw that way

### System information

- **SolveSpace version:** 3.1~3833dd02 (built a few minutes ago)
- **Operating system:** W10 22H2

### Expected behavior

From optimal to worst:
1. just split the one line at the endpoint of the second line
2. create zero-length line segment like previous versions
3. error out ("can't do that")

### Actual behavior

[image]

### Additional information

🙏〰✂

ruevs, 2023-01-06:
Yes, this is a regression. https://github.com/solvespace/solvespace/commit/b399d9a1 (January 2022) makes only two message boxes and splits the lines with a zero length edge as expected.

I bisected and this commit introduces the regression:

b48ce240c7bdcb405d9c804f81054486412ccd41 is the first bad commit
Author: @robnee
Date:   Sun Feb 7 10:10:41 2021 -0500

    Make scheduled/deferred task order deterministic

MrBlockers, 2023-01-18:
Hello, not sure if I should create a new issue, but this seems related enough. I found more reproduction cases on master 3.1 [10c6c095]. 

Make a triangle, then make a tangent arc from one of its points. When you delete one of the generated construction lines (which one or if both seems to somewhat depend on the creation order of the entities in the sketch?), then message box explosion. In my repro file I have a few other similar geometries that cause the issue. Check reproduction picture/file below. The common thread seems to be a line segment which is constrained on both ends and then has another "non trivial" constraint which shares an entity with one of the other constraints on the segment being deleted. I am having mixed success using other constraints such as symmetry or parallelism to trigger the avalanche.

I haven't finished grokking the generation code, but it seems like for whatever reason GenerateAll is failing to prune the additional "non-trivial" constraint, and looping back on itself. If nobody else is tackling this right now I'll try to hunt it down because it is getting in my way surprisingly often 😅 .

[image]
[Reproduction.zip](https://github.com/solvespace/solvespace/files/10449644/Reproduction.zip)

robnee, 2023-01-28:
I'm glad that someone is looking at this.  Seems my original "fix" for the non-deterministic timer behavior on windows caused this regression.  Sorry!

ruevs, 2023-01-29:
> Seems my original "fix" for the non-deterministic timer behavior on windows caused this regression. Sorry!

Hey, you do not need to be sorry - bugs happen - it's software :-) And you've made so many nice contributions.

What do you think about #1333? I think we may be missing another race condition somewhere.

PR #1333 (2023-02-03): Fix MessageBox Avalanches due to Message blocking
This fixes #1320, but not #1239. The root cause for the avalanche of messages is due to how the refresh timer system calls GenerateAll. When GenerateAll is called by Refresh, if a Message occurs, that Message will block GenerateAll. It _doesn't_ block subsequent calls from the timer to Refresh (presumably from a separate thread). Because the "scheduledGenerateAll" flag is not cleared until after the generation is unblocked, each following refresh triggers another call to GenerateAll. By reversing the flag clear and call, it breaks the cycle. I  [... cut]
commit d6e97091 (2023-02-03): Fix MessageBox Avalanches due to Message blocking

This fixes #1320. The root cause for the avalanche of
messages is due to how the refresh timer system calls GenerateAll. When
GenerateAll is called by Refresh, if a Message occurs, that Message will
block GenerateAll. It _doesn't_ block subsequent calls from the timer to
Refresh (presumably from a separate thread). Because the
"scheduledGenerateAll" flag is not cleared until after the generation is
unblocked, each following refresh triggers another call to GenerateAll.
By reversing the flag clear and call, it breaks the cycle. I don't think
th [... cut]
closed 2023-02-03 by phkahler