solvespace/solvespace #1324: Infinite Addition Constraints Messages

MigratingPain, 2023-01-08:
### System information
  SolveSpace Version : 3.1~3d3d5c78
  Operating System : Windows 11

### Expected behavior
  Single message box should appear.

### Actual behavior
  Infinite message boxes appear. Sometimes crashes the program 
  and other times requires task manager to kill the software.

### Additional information
  Attached video shows a simple way of recreating the problem.

https://user-images.githubusercontent.com/15020900/211180978-55f2f11b-2773-4960-9635-9f81a650f58b.mp4

MigratingPain, 2023-01-08:
Another post discussed this issue but mentioned no way of recreating the problem.

Link to other discussion
https://github.com/solvespace/solvespace/issues/1239

ruevs, 2023-01-08:
Duplicate to #1320

closed 2023-01-08 by ruevs

ruevs, 2023-02-06:
Fixed in https://github.com/solvespace/solvespace/pull/1333 https://github.com/solvespace/solvespace/commit/d6e970918f5ea77757a6d21da8b11f9c66882422