solvespace/solvespace #1325: Use N for equal angle constraints

phkahler, 2023-01-09:
### System information

- **SolveSpace version:** ALL
- **Operating system:** ALL

### Expected behavior

We recently added variadic constraints, so that multiple arc/lines/etc can be constrained to be equal at once rather than just 2 at a time. This is in conflict with the equal angle constraint which is created by selecting 3 or 4 lines and applying an equal constraint. We would like 3 or 4 lines to be made equal length just like when 2, or 5, 6, 7, etc... are selected.

### Actual behavior

When 3 or 4 lines are selected for equality, an equal angle constraint it created instead.

### Additional information

I am proposing the "N" or angle constraint be changed to create an equal angle when 3 or 4 lines are selected. This would create a more consistent behavior for the variadic "equal" constraint. The N constraint currently creates a fixed dimension (angle) between 2 lines or normals and does nothing with 3 or 4 lines. Maybe it's somewhat intuitive that the same key can also create the equal angle constraint?

phkahler, 2023-01-10:
@jwesthues  this wouldn't break compatibility would it? Constraints are not created by requests?

jwesthues, 2023-01-10:
`Constraint`s are always stored directly in the file, never generated dynamically in the manner of an `Entity` from a `Request`. If I understand correctly, you're changing only the UI, and automatically creating multiple of the same old binary equality constraints. In that case I'd expect perfect compatibility in both directions (old app / new file, and new app / old file).

closed 2023-01-11 by ruevs