solvespace/solvespace #1334: Feature Idea: Variadic "Make Arc" function

MrBlockers, 2023-01-21:
I find myself often wanting to radius a whole bunch of points on a sketch, and while you can click each vertex and press Shift+A, I feel like it is so much faster and more natural to apply it to all selected vertices at once. Especially since you can do it with constraints now.

I have already been using a small hacky patch for a while to get this feature, but I wanted to put out a feeler to see if I should polish it and contribute it. 

The only issue / UX problem I don't have an answer for is what to do when one of the corners fails to construct. Right now I only have it report the error and continue on with the rest. The Undo will need to be adjusted to undo all successfully generated arcs. 

Another small thing for consideration, it would be nice to add a "Make Arcs equal" checkbox to the tools configuration screen so that you may have an equivalence constraint generated automatically for all of the arcs. I suspect though that this will take care to not blow up a sketch in edge cases where the new arcs resolve to different sizes. 

Is there any interest in this minor usability tweak?

phkahler, 2023-01-22:
@MrBlockers  From a user point of view I like this idea. I have not looked at the code for the existing function, but there are some things I'm not a fan of. I'd like to see it implemented at least as well as the it currently is. For example, draw a rectangle and then round the 4 corners. Explode the sketch "\" to see all the lines. There will be the original 4 lines, which I think need to be preserved in case anything is constrained to them and to define where the original corners were. But each time an arc is created, an additional construction line is created, so we end up with 8 construction lines and 4 non-construction. I think knowing that both ends of a line are getting a rounded corner at once, you might be able to reduce that.

One thing I'm still thinking about in variadic constraints is allowing the user to select extra geometry and ignore it. For example an eQual constraint doesn't apply to points, so if we ignored points you'd be able to draw a polygon, lasso the entire thing and press Q to get all equal lines. Likewise it might be nice for the arc creator to ignore arcs and points, so you can select an entire sketch and round all line-line corners at once. Doing too many things this way will naturally lead people to want more cases to work so easy, and I'm not sure that's possible given the number of options constraints allow already ;-)

If you get this working, as a part two I'd consider leaving all the new arcs (and only the arcs) selected so the user can then press Q to make then all equal, since that's a common next step.  ;-)

I think you get the idea - do more of what we want, less of what we don't (extra entities in this case) and keep the code as clean as you can. Follow those rules and we'll probably take the contributions! ;-)

MrBlockers, 2023-01-22:
I hadn't considered it, but that makes a lot of sense to try and reuse as much generated construction geometry as possible. I do like the idea of not clearing the selection too- that is a lot more straight forward than a tool option. 

Yes, it would be amazing if there could be more intelligent/forgiving mapping of selection to commands. There has been multiple occasions where I wished I could just click a line and use the arc operation, or I've had to remake a selection for a constraint because I clicked some extra bogus. 

It seems doable to add that for the arc tool at least (e.g. lines reduce to compatible points in addition to points already in the selection) without it being too unexpected. If there is going to be some abstraction in the future between commands and selections maybe I should wait on trying to implement it though.

Thanks for the input, I'll give it a shot!