solvespace/solvespace #1338: Creating a pocket in a solid object must be done right the first time.

jbhatami, 2023-01-30:
### System information

- **SolveSpace version:** 
- **Operating system:** 

### Expected behavior

Create extrude on simple sketch.
New sketch in plane on the top face of that extrusion (two lines and point of corner selected).
Create a simple sketch that has some inwards offset from the edges of the first sketch/extrusion.
Create a new extrusion with the second sketch (intended to be a pocket).
Select a vertical edge and assign a dimension for the thickness of the second extrude.
Click "solid model as: difference" under the property manager. 
The 2nd extrusion's direction should change from adding to the 1st extrusion to removing material towards it instead.

### Actual behavior

if you assign a vertical dimension prior to clicking the "difference" radio button in the property manager, it will not change directions (remains additive instead of becoming subtractive).

### Additional information

phkahler, 2023-01-30:
This is not going to change. Switch it to difference before adding the distance constraint. Once you constrain it, the flip is not going to work. You can also drag the sxtrusiob prior to adding the dimension.

I've never cared for the flipping behavior even though it is often the correct thing to do. It would be poor design to change the direction of a constrained direction.

jbhatami, 2023-01-30:
The behavior of changing the radio dial prior to assigning a constrained dimension.

Whats the difference between the default distance (preview prior to adding dimension) and the constrained distance? Why can one flip and the other cannot.

phkahler, 2023-01-30:
The flip seems to be a convenience based on the user probably wanting a difference and union to go opposite directions. That is not a certainty but a reasonable guess. A constraint is explicitly what the user has indicated they want.

closed 2023-01-30 by phkahler

ruevs, 2023-01-30:
@jbhatami @phkahler to "flip" the direction of an extrusion enter a negative number for the distance constraint e.g. "-3". This is documented here https://solvespace.com/ref.pl#Distance.

jbhatami, 2023-01-30:
> @jbhatami @phkahler to "flip" the direction of an extrusion enter a negative number for the distance constraint e.g. "-3". This is documented here https://solvespace.com/ref.pl#Distance.

I did try that, it did not work. It appears you have to get it right in the beginning or delete and re-create the extrude feature.

phkahler, 2023-01-30:
@jbhatami  You can delete the distance constraint, drag the feature to the other side with the mouse, and then reapply the distance constraint.

phkahler, 2023-01-30:
@ruevs  I almost always select a line to set the distance, so it doesn't work. BTW I never knew about entering a negative value to flip it, people have asked for that.

@jbhatami  another option, if your distance isn't very large (or I suppose if you zoom out) is to grab the end of the extrusion with the mouse and give it a quick yank in the other direction. Sometimes this is enough to flip the direction ;-)  it has been reported as a bug, but without signed distance constraints it's not really something we can fix. It might be useful to you in this case!

ruevs, 2023-01-30:
And one more way - probably the best.

- Enable "relax constraints and dimensions" 
[image]
- Drag freely to the desired side
- Disable the option. The extrusion is snapped back to the original value of the constraint closet to the position where it is dragged.