solvespace/solvespace #1339: Scroll wheel no longer zooms

dannystaple, 2023-01-30:
### System information

- **SolveSpace version:** Version 3.1 (3.1~70bde63c)
- **Operating system:** macOS Monterey 12.6 (21G115)

### Expected behavior

Scroll wheel should be zooming (as it was in the previous releases)

### Actual behavior

The part is scroll up/down, and the horizontal middle is being placed at the mouse cursor.

### Additional information

This appears to be on multiple files. I was able to reproduce this with https://solvespace.com/2d.pl.

dannystaple, 2023-01-30:
No sketch needed - can be reproduced just showing the default planes.
I can see this in:
- Version 3.0 (3.0~0e0b0252)
- Version 3.0 (3.0~bcb8cd2c)
And older.  am going to reboot - in case this is me.

dannystaple, 2023-01-30:
This persisted across reboots. And is seen in older versions. I am suspecting some solvespace setting that I've not understood here.

ruevs, 2023-02-12:
In my virtual machine with macOS Mojave (10.14.6) under VirtualBox on Windows the scroll wheel still zooms just fine. I am on https://github.com/solvespace/solvespace/commit/d6e970918f5ea77757a6d21da8b11f9c66882422 built from source.

Perhaps this commit https://github.com/solvespace/solvespace/commit/cf4defcd474e0a55cfb5970265b3d1f0e99d16b0 makes a difference? @nanoant @vespakoen ?

nanoant, 2023-02-12:
@dannystaple Hi. Can you please let me know what kind of mouse you are using?

ruevs, 2023-02-15:
A generic three button (two buttons and a clicky scroll wheel) PC USB HID mouse. As I said - my macOS is a virtual machine under Windows.

dannystaple, 2023-02-15:
What is interesting is this HAD worked - my suspicion is that I've managed to modify some solvespace related setting that has changed it's behaviour, since I was able to reproduce it in far earlier versions.
The mouse I am using is a Logitech MX Ergo trackball, connected via Bluetooth.

dannystaple, 2023-02-15:
Is there a config/settings/preferences file somewhere I could blow away in case that is where this is being stored?

RM250J, 2023-02-17:
I just came across this issue with one of my Logitech mice (M185 w/ nano receiver), I installed Solvespace on a Pinebook Pro from source as it wasn't in Fedora repos on Arm64. So I tried my Logitech Pebble with nano receiver and it works. Here's where it gets weird, I installed Solvespace on a RPi400 64bit to test and the M185 scroll wheel works on that! The M185 scroll wheel issue is also present on an x86 64bit install of Ubuntu

dannystaple, 2023-02-17:
Hello, I tried the edge build out, and no effect. I also realise that I probably changes mouse, as I was using a cheaper Jellycomb trackball for a while before getting a fresh MX Ergo (the buttons go on these sadly). I'll try out switching to the Jellycomb and tell you if that works.

nanoant, 2023-02-17:
@dannystaple, @ruevs et all,

Just to make sure we understand the issue and its nature... there are two camps, folks using mouse and a wheel that want to have a wheel acting as a zoom and folks using touch pad that want to make it pan the view.
Unfortunately in both cases these things are mapped to the same `- (void)scrollWheel:(NSEvent *)nsEvent` on macOS.

Since recent changes for touch pad guys broke Apple Magic Mouse zoom, I did a lot of effort in cf4defcd474e0a55cfb5970265b3d1f0e99d16b0 to distinguish trackpads from mouse wheels including Apple Magic Mouse that I use myself most of the time.

I have 3+ mice around and Mac mini running macOS 13 + MacBook Pro running BigSur and the solution works well with all of them and the trackpad. But am aware it may be not perfect and probably it is impossible to make it work in all cases including macOS being virtualised with physical mouse mapped to I have no clue what by VMware, maybe trackpad!?

Therefore @ruevs the only solid way to solve this once and for all is to have a new setting:
`Scroll/track behavior: [x] Auto (default/current impl.) [ ] Pan [ ] Zoom`

I know you did not want to have such setting, but I guess maybe we need one. Thoughts?

[image]

dannystaple, 2023-02-19:
I personally like the idea of a setting that fixes such behaviour.