solvespace/solvespace #1349: Web solvespace crashes (`allocate` is a library symbol and not included by default)

jkrei0, 2023-03-02:
### System information

- **SolveSpace version:** 25b5977 (latest) 
- **Operating system:** WSL Ubuntu (and Firefox on Windows) 

### Expected behavior
The web version does not crash after reloading.

### Actual behavior
SolveSpace compiles fine, then loads once.
Upon reloading the page, SolveSpace crashes after loading (sometimes it crashes the first time, with the same error message). Clearing site data and reloading again fixes the problem. (until the page is reloaded again - then you must clear site data again).

### Additional information
Console output
```
Status: Downloading... solvespace.html:1:754
Status: Preparing... (0/1) solvespace.html:1:754
Status: Downloading data... (131072/2324429) solvespace.html:1:754
Status: Downloading data... (2324429/2324429) solvespace.html:1:754
Status: Preparing... (0/1) solvespace.html:1:754
... (omitted "Preparing..." logs)
Status: Preparing... (90/91) solvespace.html:1:754
Status: Running... solvespace.html:1:754
SolveSpace! solvespace.js:1:42393
<empty string> solvespace.js:1:42393
`allocate` is a library symbol and not included by default; add it to your library.js __deps or to DEFAULT_LIBRARY_FUNCS_TO_INCLUDE on the command line (e.g. -sDEFAULT_LIBRARY_FUNCS_TO_INCLUDE=$allocate) solvespace.js:1:34960
    warnOnce http://localhost:8088/bin/solvespace.js:1
    get http://localhost:8088/bin/solvespace.js:1
    782829 http://localhost:8088/bin/solvespace.js:1
    runEmAsmFunction http://localhost:8088/bin/solvespace.js:1
    _emscripten_asm_const_int http://localhost:8088/bin/solvespace.js:1
    x http://localhost:8088/bin/solvespace.js:1
    <anonymous> http://localhost:8088/bin/solvespace.wasm:1173852
    <anonymous> http://localhost:8088/bin/solvespace.wasm:1173396
    <anonymous> http://localhost:8088/bin/solvespace.wasm:1048356
    x http://localhost:8088/bin/solvespace.js:1
    createExportWrapper http://localhost:8088/bin/solvespace.js:1
    callMain http://localhost:8088/bin/solvespace.js:1
    doRun http://localhost:8088/bin/solvespace.js:1
    run http://localhost:8088/bin/solvespace.js:1
`ALLOC_NORMAL` is a library symbol and not included by default; add it to your library.js __deps or to DEFAULT_LIBRARY_FUNCS_TO_INCLUDE on the command line (e.g. -sDEFAULT_LIBRARY_FUNCS_TO_INCLUDE=$ALLOC_NORMAL) solvespace.js:1:34960
Uncaught TypeError: allocate is not a function
    782829 http://localhost:8088/bin/solvespace.js:1
    runEmAsmFunction http://localhost:8088/bin/solvespace.js:1
    _emscripten_asm_const_int http://localhost:8088/bin/solvespace.js:1
    x http://localhost:8088/bin/solvespace.js:1
    x http://localhost:8088/bin/solvespace.js:1
    createExportWrapper http://localhost:8088/bin/solvespace.js:1
    callMain http://localhost:8088/bin/solvespace.js:1
    doRun http://localhost:8088/bin/solvespace.js:1
    run http://localhost:8088/bin/solvespace.js:1
solvespace.js:1:34059

```

ruevs, 2023-09-06:
@jkrei0 I can not reproduce this by compiling locally but "natively" on Windows (not under WSL).
Neither does it happen with the version from @verylowfreq
https://verylowfreq.github.io/experimental-solvespace-on-browser/
Does this one work for you?

jkrei0, 2023-09-09:
Interesting ... @verylowfreq's version works fine for me.

Perhaps I'm doing something wrong, but I would have no idea what. I built it again following the instructions in the readme, and it still is broken for some reason. (you can reproduce here: https://jkrei0.github.io/solvespace-bin-36ecb85b/solvespace.html).

> I can not reproduce this by compiling locally but "natively" on Windows (not under WSL).

Is it possible to compile natively on windows? I've not gotten it to work, but that's probably due to my own inabilities.

ruevs, 2023-09-09:
> you can reproduce here: https://jkrei0.github.io/solvespace-bin-36ecb85b/solvespace.html).

Indeed - exactly as you described. Works the first time and crashes afterwards. Clearing cookies and site data allows one more successfull load. And this is on Firefox on Android :-)

I can only guess something goes wrong with the settings, which are stored in the persistent storage (if I'm not mistaken) with your build.

As for compiling the Emscripten version under Windows "natively" - this is the only way I did it. I don't have WSL(2). The process is subtly different from the Linux instructions. When I'm on the machine I used, I'll post the notes I took here.

ruevs, 2023-09-09:
Yes the solver example you compiled does not crash - no matter how many reloads:
https://jkrei0.github.io/solvespace-bin-36ecb85b/CDemo.html
and neither does the benchmark (but it does not do much)
https://jkrei0.github.io/solvespace-bin-36ecb85b/solvespace-benchmark.html

jkrei0, 2023-09-09:
I have two other things I forget to mention,
- I also tried compiling it in a clean linux VM (not WSL), and got the same results, so maybe there's some step/package/config that you have (from working on solvespace or something else in the past) that I am missing?
- Entering data (changing a dimension or renaming a group) also crashes my build, but works on the one you linked.

verylowfreq, 2023-09-10:
@ruevs @jkrei0
The crash on reloading was reproduced.
And the following fix solved it. (With latest emsdk and solvespace master as described the following)

 - Add this option to `emcmake`: `-DCMAKE_CXX_FLAGS="-sDEFAULT_LIBRARY_FUNCS_TO_INCLUDE=['\$allocate','\$ALLOC_NORMAL']"`

(So the full commandline is `emcmake cmake .. -DCMAKE_BUILD_TYPE=Release -DENABLE_LTO="ON" -DENABLE_TESTS="OFF" -DENABLE_CLI="OFF" -DENABLE_COVERAGE="OFF" -DCMAKE_CXX_FLAGS="-sDEFAULT_LIBRARY_FUNCS_TO_INCLUDE=['\$allocate','\$ALLOC_NORMAL']"`)

Information:
 - Machine: WSL2 on Windows 11; Ubuntu Linux 22.04.2
 - solvespace: 36ecb85bb381235521f7e0e1b913d65308878f48
 - emsdk: 'latest' '3.1.45' 'sdk-releases-2b7c5fb8ffeac3315deb1f82ab7bf8da544f84a1-64bit'

The update of Emscripten in Sep,2022 looks causing this issue.
https://github.com/emscripten-core/emscripten/blob/main/ChangeLog.md#3121---09092022
> The LEGACY_RUNTIME setting is no longer enabled by default. If you use any of these legacy runtime functions (except in library code with explicit dependencies) then you would need to set LEGACY_RUNTIME on the command line or add the ones you need to DEFAULT_LIBRARY_FUNCS_TO_INCLUDE:
>  - (function name list...)
>  - allocate

jkrei0, 2023-09-12:
Yep, this fixes it! I was able to build a working copy of the latest SolveSpace.