solvespace/solvespace #1351: Crash on launch

zohozer, 2023-03-17:
### System information

- **SolveSpace version:** 3.1
- **Operating system:** SteamOS 3.4

### Expected behavior

I do expect to be able to open the program.

### Actual behavior

It is crashing on launch so I can't open the program

### Additional information

I do tried to launch-it from the Terminal app and I do get the following error:

~> com.solvespace.SolveSpace 
Sorry, only UTF-8 locales are supported.

zohozer, 2023-03-17:
If I do run the `locale` comand I get this result:

~> locale
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME=ro_RO.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=en_GB.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT=ro_RO.UTF-8
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=

zohozer, 2023-03-18:
Ok, found a working solution here: https://solvespace.com/forum.pl?action=viewthread&parent=1649

~> LC_ALL=en_US.UTF-8 com.solvespace.SolveSpace

ruevs, 2023-03-18:
Shall we close this than?

closed 2023-03-23 by zohozer