solvespace/solvespace #1354: Large Dimensions result in Incompatible Constraints

zegelin, 2023-03-27:
Trying to model large systems that have dimensions in metres causes the certain constraint types to fail with an "Incompatible Constraints" error. For example, two 4 metre long lines cannot be constrained perpendicular to each other, as shown in the screenshot below.

(I have View -> Dimension Units set to "Dimensions in Meters".)

[image]

Are there dimensional limits to the solver? E.g. keep everything smaller than X mm?
A work around is to model everything at a smaller scale and export with a scale factor, but it would be nice to have the model match the project scale.

### System information

- **SolveSpace version:** 3.1
- **Operating system:** Windows 10

### Expected behavior

Requests and their constraints spanning large distances should work.

### Actual behavior

Requests and their constraints spanning large distances fail with an "Incompatible Constraints" error.

### Additional information

[Incompatible Constraints Bug.zip](https://github.com/solvespace/solvespace/files/11074149/Incompatible.Constraints.Bug.zip)

ruevs, 2023-03-27:
That's an interesting bug in the solver. Thank you for the minimal reproduction model.

The interesting part is that if one deletes the constraint "c002-pt-on-line" and recreates it then it works irrespective of how long one makes the line segments. In other words if the perpendicular constraint is created first and the point on line after that it is OK, Otherwise it breaks.

Alternatively if **one** of the dimensions is decreased to 1851mm or less it works... and breaks again of it goes above.

This needs to be debugged. Useful code https://github.com/solvespace/solvespace/issues/1247#issuecomment-1147481539

ruevs, 2024-02-07:
#1105 for cross reference.

zwn, 2024-02-09:
Another way the problem can be recreated is by drawing 10m x 10m x 2m box as a part, importing it into an an empty file and constrain any point on origin.

> I rebased https://github.com/Evil-Spirit/solvespace-master/commits/solver-debug onto master here: https://github.com/ruevs/solvespace/tree/EvilSpirit_solver-debug it is interesting to watch the solve fail, and see the equations but no immediate help. I need to think/debug much more...
> 
> #226 is the original issue for which this code was created.

I tried the code on my failing case:

- [Debug / Equations][1]
- [Debug / Folded Equations][2]

But `Debug / Show Solve Way` is not showing any blue lines. I've seen them only once but I am not able to recreate it anymore.

[image]

Are these of any help?

[1]: http://www.texrendr.com/?eqn=\left(p80030003^{2}%20%2B%20p80030004^{2}-p80030005^{2}-p80030006^{2}\right)\cdot%20-5000%20%2B%20\left(2\cdot%20p80030004\cdot%20p80030005-2\cdot%20p80030003\cdot%20p80030006\right)\cdot%20-5000%20%2B%20\left(2\cdot%20p80030003\cdot%20p80030005%20%2B%202\cdot%20p80030004\cdot%20p80030006\right)\cdot%200%20%2B%20p80030000-0\\%20\left(2\cdot%20p80030003\cdot%20p80030006%20%2B%202\cdot%20p80030004\cdot%20p80030005\right)\cdot%20-5000%20%2B%20\left(p80030003^{2}-p80030004^{2}%20%2B%20p80030005^{2}-p80030006^{2}\right)\cdot%20-5000%20%2B%20\left(2\cdot%20p80030005\cdot%20p80030006-2\cdot%20p80030003\cdot%20p80030004\right)\cdot%200%20%2B%20p80030001-0\\%20\left(2\cdot%20p80030004\cdot%20p80030006-2\cdot%20p80030003\cdot%20p80030005\right)\cdot%20-5000%20%2B%20\left(2\cdot%20p80030003\cdot%20p80030004%20%2B%202\cdot%20p80030005\cdot%20p80030006\right)\cdot%20-5000%20%2B%20\left(p80030003^{2}-p80030004^{2}-p80030005^{2}%20%2B%20p80030006^{2}\right)\cdot%200%20%2B%20p80030002-0\\%20\sqrt{p80030003^{2}%20%2B%20p80030004^{2}%20%2B%20p80030005^{2}%20%2B%20p80030006^{2}}-1\\

[2]: http://www.texrendr.com/?eqn=\left(p80030003^{2}%20%2B%20p80030004^{2}-p80030005^{2}-p80030006^{2}\right)\cdot%20-5000%20%2B%20\left(2\cdot%20p80030004\cdot%20p80030005-2\cdot%20p80030003\cdot%20p80030006\right)\cdot%20-5000%20%2B%20p80030000-0\\%20\left(2\cdot%20p80030003\cdot%20p80030006%20%2B%202\cdot%20p80030004\cdot%20p80030005\right)\cdot%20-5000%20%2B%20\left(p80030003^{2}-p80030004^{2}%20%2B%20p80030005^{2}-p80030006^{2}\right)\cdot%20-5000%20%2B%20p80030001-0\\%20\left(2\cdot%20p80030004\cdot%20p80030006-2\cdot%20p80030003\cdot%20p80030005\right)\cdot%20-5000%20%2B%20\left(2\cdot%20p80030003\cdot%20p80030004%20%2B%202\cdot%20p80030005\cdot%20p80030006\right)\cdot%20-5000%20%2B%20p80030002-0\\%20\sqrt{p80030003^{2}%20%2B%20p80030004^{2}%20%2B%20p80030005^{2}%20%2B%20p80030006^{2}}-1\\

zwn, 2024-02-16:
Cadquery uses scale presumably to make the process scale invariant. The scale used there is the [diagonal of the bounding box](https://github.com/CadQuery/cadquery/blob/551a23157026faef423e7cb7caaf6f84da91a562/cadquery/assembly.py#L429).