solvespace/solvespace #1363: Solvespace segfaults on routine operations

BenLand100, 2023-04-23:
I've highlighted one specific scenario here, where drawing circles and adding dimension constraints consistently results in a segfault,  which hard crashes the program. I raise this issue because this is an absurdly common failure mode in Solvespace, and has been persistent for many versions including the most recent. What's going on here? Why are segfaults as a result of routine user input even possible in Solvespace?

### System information

- **SolveSpace version:** 3.1~70bde63c 
- **Operating system:** Gentoo

### Expected behavior

Anything besides hard crash.

### Actual behavior

Segmentation fault

### Additional information

1. Open a new project
2. New Group -> Link/Assemble -> Choose STL to import
3. Create a new workplane via any method
4. Draw some circles and try to add dimension constraints.
5. I never make it past 3 circles until Solvespace crashes with a segfault

Sure maybe my STL is to blame in this specific case (I didn't create it, so can't really say), but _why is it causing a segmentation fault_???

```
Generate::DIRTY took 385 ms
Vector::WithMagnitude(-2.5) of zero vector!
Vector::WithMagnitude(-2.5) of zero vector!
Vector::WithMagnitude(-2.5) of zero vector!
Vector::WithMagnitude(-2.5) of zero vector!
Vector::WithMagnitude(-2.5) of zero vector!
Vector::WithMagnitude(-2.5) of zero vector!
Vector::WithMagnitude(-2.5) of zero vector!
Vector::WithMagnitude(-2.5) of zero vector!
Vector::WithMagnitude(-2.5) of zero vector!
Vector::WithMagnitude(-14.972) of zero vector!
Segmentation fault
```

phkahler, 2023-05-23:
I could not reproduce this on Fedora. @BenLand100 can you provide a test case?

ruevs, 2023-05-31:
I can not reproduce it either:

[image]

I sketched in a workplane in the same group as the STL, in a workplane in a separate group and in 3d in a separate group.

This is using https://github.com/solvespace/solvespace/issues/1085#issue-963322756 
https://github.com/solvespace/solvespace/files/6950016/blender_cube_monkey_stl.zip

ruevs, 2024-08-18:
@BenLand100 ? Can you point us to how to reproduce this? More details of your system? You claim that this "has been persistent for many versions" and it seems reproducible, so it would be nice to catch ot and fix it.

BenLand100, 2024-08-18:
@ruevs I don't recall what STLs I was testing with before, so I picked a random recent one and repeated the procedure I described above. Seems I'm still on the same solvespace version as the issue reports, and it is still reproducible. The stdout is below. No longer a segmentation fault, but a bit more detail shows up. Typical options such as `-h` `--help` `-v` etc don't seem to work with `solvespace`, so I'm not really sure how to proceed for next steps to get more verbosity or whatnot, but happy do perform any debugging steps you can describe. I use FreeCAD now, so my frustration level is significantly reduced from the initial report.

```
benland100@blandtoo ~ $ solvespace
SolveSpace!

connect failed: No such file or directory

Link STL triangle mesh.
22268 triangles
11132 vertices
Vector::WithMagnitude(-34.7003) of zero vector!
Vector::WithMagnitude(-34.7003) of zero vector!
Vector::WithMagnitude(-34.7003) of zero vector!
Vector::WithMagnitude(-34.7003) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
[... 1099 lines cut ...]
Vector::WithMagnitude(-0.5) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-0.5) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-0.5) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-0.5) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-0.5) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-0.5) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-0.5) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-0.5) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-0.5) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-0.5) of zero vector!
Vector::WithMagnitude(-2) of zero vector!
Vector::WithMagnitude(-0.5) of zero vector!
mimalloc: assertion failed: at "/var/tmp/portage/dev-libs/mimalloc-2.1.2/work/mimalloc-2.1.2/src/alloc.c":267, mi_page_usable_size_of
  assertion: "ok"
Aborted
```

BenLand100, 2024-08-18:
I'd also like to clarify the highlight in the OP that the "1) load STL 2) draw circles with constraints" procedure is just particularly reliable in causing a crash. I used to get similar crashes with solvespace frequently enough to be trained to save before any moderate operations. Most projects reached a point where it was just not possible to make further changes with the tedium of having to restart and redo work. I don't think this is anything particular to drawing circles or the STL in use.

BenLand100, 2024-08-18:
System Details: i9-12900k; 128GB Memory; Linux 6.10.5; RTX 3090; nvidia-drivers 555.58.02; plenty of disk space. 

```
benland100@blandtoo ~ $ equery g solvespace
 * Searching for solvespace ...

 * dependency graph for media-gfx/solvespace-3.1-r1
 `--  media-gfx/solvespace-3.1-r1  amd64 
   `--  dev-cpp/atkmm-2.28.3  (dev-cpp/atkmm) amd64 
   `--  dev-cpp/glibmm-2.66.7  (dev-cpp/glibmm) amd64 
   `--  dev-cpp/gtkmm-3.24.9  (dev-cpp/gtkmm) amd64  [X]
   `--  dev-cpp/pangomm-2.46.3  (dev-cpp/pangomm) amd64 
   `--  dev-libs/glib-2.78.6  (dev-libs/glib) amd64 
   `--  dev-libs/json-c-0.17  (dev-libs/json-c) amd64 
   `--  dev-libs/libsigc++-2.12.0  (dev-libs/libsigc++) amd64 
   `--  dev-libs/libspnav-0.2.3-r1  (dev-libs/libspnav) amd64  [X]
   `--  media-libs/fontconfig-2.15.0  (media-libs/fontconfig) ~amd64 
   `--  media-libs/freetype-2.13.3  (media-libs/freetype) ~amd64  [X]
   `--  media-libs/libpng-1.6.43  (media-libs/libpng) amd64 
   `--  sys-libs/zlib-1.3.1-r1  (sys-libs/zlib) ~amd64 
   `--  virtual/opengl-7.0-r2  (virtual/opengl) amd64 
   `--  x11-libs/cairo-1.18.0  (x11-libs/cairo) amd64  [X]
   `--  x11-libs/gtk+-3.24.41-r1  (x11-libs/gtk+) amd64  [X]
   `--  dev-libs/mimalloc-2.1.2  (dev-libs/mimalloc) amd64 
   `--  dev-cpp/eigen-3.4.0-r1  (dev-cpp/eigen) amd64 
   `--  virtual/pkgconfig-3  (virtual/pkgconfig) amd64 
   `--  app-alternatives/ninja-1  (app-alternatives/ninja) amd64 
   `--  dev-build/cmake-3.28.5  (>=dev-build/cmake-3.20.5) amd64 
```

Let me know any details that might help

ruevs, 2024-08-18:
@BenLand100 would you be willing to share an STL that causes the problem here?

BenLand100, 2024-08-18:
Here's the file I tried today: [Coma Adapter SonyE-M48x0.75.stl.zip](https://github.com/user-attachments/files/16651933/Coma.Adapter.SonyE-M48x0.75.stl.zip)

ruevs, 2024-08-19:
Interesting. On Windows I do not get any of the `Vector::WithMagnitude(x) of zero vector!` warnings and also I do not get
```
mimalloc: assertion failed: at "/var/tmp/portage/dev-libs/mimalloc-2.1.2/work/mimalloc-2.1.2/src/alloc.c":267, mi_page_usable_size_of
  assertion: "ok"
```

I also tried on the official 3.1 release - work as well.

This seems to be something specific to your system since we rarely get reports of crashing. Your version "https://github.com/solvespace/solvespace/commit/70bde63c" is exactly the 3.1 release. How did you install it? Or did you compile from source?

It seems someone with more experience on (and using) Linux will need to help you.

https://github.com/user-attachments/assets/61a13f03-37ed-4466-9229-a48baf819019

ruevs, 2024-08-19:
Even though your mimalloc version is newer the assert is here:
https://github.com/microsoft/mimalloc/blob/f819dbb4e4813fab464aee16770f39f11476bfea/src/alloc.c#L231
However on Windows this is not defined:
https://github.com/microsoft/mimalloc/blob/f819dbb4e4813fab464aee16770f39f11476bfea/src/alloc.c#L218
so the version without an assert is used:
https://github.com/microsoft/mimalloc/blob/f819dbb4e4813fab464aee16770f39f11476bfea/src/alloc.c#L285

ruevs, 2024-08-20:
I re-tested with a debug build where the assert in mimalloc is enabled and it did _not_ trigger. Also ran with [address sanitizer](https://learn.microsoft.com/en-us/cpp/sanitizers/asan) enabled - no findings (quite a bit of effort wend into on not leaking and corrupting memory over the years).

Does it crash for you if you just open "Hmmmm.slvs" from this: [ComaAdapterSony_Linked.zip](https://github.com/user-attachments/files/16677176/ComaAdapterSony_Linked.zip) ?

Would you be willing to [compile](https://github.com/solvespace/solvespace?tab=readme-ov-file#building-for-linux) a debug version and try to [debug it to catch a call stack](https://github.com/solvespace/solvespace/blob/master/CONTRIBUTING.md#debugging-with-gdb)?

ruevs, 2024-11-28:
@BenLand100 are you still (trying to) use SolveSpace? Does it still crash? If not - did you find out what was the problem?