solvespace/solvespace #1366: Feature Request: New constraint "As Cubic Bezier"  for a 5 points spline

xmarduel, 2023-05-03:
### System information

- **SolveSpace version:** 
- **Operating system:** 

### Expected behavior

It is actually possible to define the smallest spline with 5 control points.

It would be nice to have a new constraint on such a 5 control points spline named "As Cubic Bezier"
which would simply force the mid point of the spline to lie exactly on (the middle of) the Cubic Bezier
defined through the start/end points and the two other control points for the tangents.

With this constraint, the SVG export would also know that only 1 cubic bezier has to be exported (instead of
actually 2 for the 5 points spline).

With this constraint, moving one of the remaining 4 control points would move the mid point automatically
so that it stays on the implicit bezier curve (moving the mid point would change in some way the others
2 control points, but this seems to be more tricky to achieve).

Wouldn't it be a good idea/feature ?

It is easier to achieve symmetry with the cubic bezier curve as with the actual 5 points spline I believe.

Best regards,
XM

### Actual behavior

cannot constraint a spline to a cubic bezier curve

### Additional information

PS: A constraint of the 5 points spline "As Quadratic Bezier" could also be possible, but the mid point
of the 5 points spline lies on the curve, while the control point of quadratic bezier does not, so this
is a little bit different. In this case, the mid point of the spline would play the role of the quadratic
bezier curve mid control point, the curve would not pass through this control point, and the 2nd and 4th 
control points of the spline would be derived from the 3 other points. 
Of course, SVG export should thus write a single Q item in the path.

PPS: I do not know a single CAD program which does SVG path output correctly excepted SolveSpace, it is amazing.

phkahler, 2023-05-04:
All sketch entities in Solvespace are composed of 1 or more nurbs curves of degree 3 or lower. Splines of higher order than 3 are composed of multiple (I think 2nd order bezier) splines with automatic constraints on most of the control points (all but 2). I just want to clarify in the hope it will help you achieve your goal.

BTW given that simple representation of all entities, it's easy to do SVG export  ;-)

xmarduel, 2023-05-04:
Ouups sorry, it is very possible to generate a cubic bezier. Great! I had always clicked 1 time too often on the screen! Sorry for the garbage.
SIncerely
XM