solvespace/solvespace #1374: Entities outside of the selection box get selected

liuxilu, 2023-06-01:
### System information

- **SolveSpace version:** 3.1~659215d8
- **Operating system:** Win10

### Expected behavior
Only the horizontal line is selected by doing this
[image]

### Actual behavior
The two lines are selected.
### Additional information
Because only AABB is used for detection, that's rough
https://github.com/solvespace/solvespace/blob/fd25424ab5e5a1c4158c1e65b33b6568b5f9bc2b/src/draw.cpp#L220-L224

liuxilu, 2023-09-23:
Why closed?

vespakoen, 2023-09-23:
I thought that the conclusion was that a more precise selection method would take too many resources, and would make things too slow, but thinking about it now there might be a possibility to make an exceptions for segments.

We could check if the selection box "lines" cross with segments, in stead of checking if the boxes overlap.

However, that would still not fix selection of arcs and splines, but could be an improvement overall that doesn't take too many resources?

[image]

So we would basically make an exception for the above only, but would still have "odd" selection for the below:

[image]

Would that be a good middle ground?

ruevs, 2023-09-23:
That's a reasonable sounding possibility. It will require quite a few extra calculations but will still be a linear time algorithm (on the number of line segments). Could be acceptably fast.

phkahler, 2023-09-23:
@vespakoen   Do both.  Use the existing fast but sloppy bounding box method, and then use a more precise method just for things that pass the coarse test. Entities have to pass both to get selected.

ruevs, 2023-09-23:
> Use the existing fast but sloppy bounding box method, and then use a more precise method just for things that pass the coarse test.

Obviously... The line intersection test should be  just to cull the selection list made with bounding boxes. Doing anything else would be rather... strange/unwise...

liuxilu, 2023-09-24:
How about requiring one point of the object to be in the box?
One-point-selected can be the indicator for user (my) interest, I think.

ruevs, 2023-09-24:
> How about requiring one point of the object to be in the box?

Definitely not! This is worse than even the current behaviour (before the fix).

phkahler, 2023-09-24:
Some tools will select anything overlapping the box if you draw the box left-to-right and only things completely inside the box when drawn right-to-left.

I thought solvespace was trying to do this but maybe not. Maybe it should. ;-)

ruevs, 2023-09-24:
> Some tools will select anything overlapping the box if you draw the box left-to-right and only things completely inside the box when drawn right-to-left.

Interesting! I never knew this existed - in any software.

> I thought solvespace was trying to do this but maybe not. Maybe it should. ;-)

This could be a pretty useful "power user" feature. I wonder how widespread it is? Apparently AutoCAD uses it https://forum.vectorworks.net/index.php?/topic/42974-marquee-select-like-autocad/  and it seems Adobe Illustrator does not :-) https://illustrator.uservoice.com/forums/333657-illustrator-desktop-feature-requests/suggestions/31863601-select-inside-marquee-window

And an example of the behaviour on the web
https://playground.babylonjs.com/#QHHXSD#1

Interesting...

PR #1411 (2023-10-14): Make segment selection more precise
Fixes #1374
closed 2023-10-14 by ruevs