solvespace/solvespace #1375: Crash on IDF import with duplicate drilled holes

matscalia, 2023-06-02:
### System information

- **SolveSpace version:** 3.1~70bde63c
- **Operating system:** Ubuntu 23.04

### Expected behavior

Ceate new group with pcb or notify of duplicate drilled holes and abort linking.

### Actual behavior

Segmentation fault in importidf.cpp calling `SShell::MakeFromExtrusionOf`.
`sblss.l.elem` is actually a null pointer when there are multiple lines in `.DRILLED_HOLES` sections with the same diameter, x and y.
After I removed data duplication with a C++ program all seems to work fine.

### Steps to reproduce

Clone [The Card](https://github.com/benvallack/The-Card), open KiCad6.0.11+dfsg-1, select `The Card Tears Rounded.kicad_pcb`, then File->Export->IDFv3. Link *.emn file.

### Additional information

I genuinely do not know if the solution belongs to kicad, solvespace or design practices. I think that solvespace should at least notify the user though.

phkahler, 2023-06-02:
It never occurred to me that a hole might be duplicated. I'm tempted to say it's a KiCAD issue for doing that, but solvespace shouldn't crash either. If we could detect it we could ignore it.

Related, I have no idea how many people use the idf feature. If there is a lot of interest I could add the part models, but it's a bunch of work.

phkahler, 2023-06-02:
@matscalia  can you attach the emn file that causes the crash? I need to see how it's structured. There is only one call to that function and there is an interesting comment just before its called:

https://github.com/solvespace/solvespace/blob/fd25424ab5e5a1c4158c1e65b33b6568b5f9bc2b/src/importidf.cpp#L511

matscalia, 2023-06-03:
I asked the author of the pcb if I could post the emn file given there isn't an explicit license in the repository I linked. Have you tried following the steps to reproduce the issue? As soon as Ben Vallack agrees, I'll post the emn file here.

I am also wrote a fix in order to not insert duplicate entities from `EntityList *el`. Searching the entity list gets quite slower though. A faster approach is to create a struct with just holes data, add operators, add hash specialization and keep an `std::unordered_set` of seen ones. As soon as I check again "How to contribute" just to be sure to have not messed something up I'll make a pull request for the in place entity list search. I mean, if you think this is the right approach.

As for the part models, I'm not particularly interested at the moment even if it would be a nice touch.