solvespace/solvespace #1376: Linux: Crash on launch (repeat of closed issue #1351)

Neinei0k, 2023-06-02:
### System information

- **SolveSpace version:** 3.1 Flatpak
- **Operating system:** Linux Mint 20.3

### Expected behavior

Open program.

### Actual behavior

Crash on launch.

### Additional information
Repeat issue #1351.
The original issue was closed, because a workaround was found (manually setting LC_ALL environment variable). But this is only a temporary solution, so the original issue should not have been closed. I found it because I have the same problem. Setting LC_ALL also fixed it.

But the issue #1351 was probably caused by LC_TIME and LC_MEASUREMENT being set to ro_RO.UTF-8. I have the same problem because my LC_TIME is set to ja_JP.UTF-8. If I set LC_TIME to en_US.UTF-8 or en_GB.UTF-8, the program runs, but when I try other values, including ro_RO.UTF-8, it crashes. I've tried setting LC_MEASUREMENT to ro_RO.UTF-8 while LC_TIME is en_US.UTF-8 and the program also crashed.

My output of locale command
```
LANG=en_US.UTF-8
LANGUAGE=en_US
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=en_GB.UTF-8
LC_TIME=ja_JP.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=en_GB.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=en_GB.UTF-8
LC_NAME=en_GB.UTF-8
LC_ADDRESS=en_GB.UTF-8
LC_TELEPHONE=en_GB.UTF-8
LC_MEASUREMENT=en_GB.UTF-8
LC_IDENTIFICATION=en_GB.UTF-8
LC_ALL=
```
For testing purposes, I've used the following command to change environment variables and print the values that flatpak sets when running solvespace
```
LC_TIME=en_GB.UTF-8 LC_MEASUREMENT=ro_RO.UTF-8 flatpak -vv run com.solvespace.SolveSpace |& grep -A 1 -e "LC_" -e "LANG"
```
The output of this particular command on my machine (the program crashed because LC_MEASUREMENT is set to ro_RO.UTF-8) is
```
F:     GDM_LANG
F:     en_US
--
F:     LANG
F:     en_US.UTF-8
--
F:     LANGUAGE
F:     en_US
--
F:     LC_ADDRESS
F:     en_GB.UTF-8
--
F:     LC_IDENTIFICATION
F:     en_GB.UTF-8
--
F:     LC_MEASUREMENT
F:     ro_RO.UTF-8
--
F:     LC_MONETARY
F:     en_GB.UTF-8
--
F:     LC_NAME
F:     en_GB.UTF-8
--
F:     LC_NUMERIC
F:     en_GB.UTF-8
--
F:     LC_PAPER
F:     en_GB.UTF-8
--
F:     LC_TELEPHONE
F:     en_GB.UTF-8
--
F:     LC_TIME
F:     en_GB.UTF-8
```

I've also tried other LC_ variables and LANG variable, if any of them is set to ro_RO.UTF-8 or ja_JP.UTF-8 (probably a lot of other values but I checked only those two), the program crashes. Changing GDM_LANG or LANGUAGE does cause the crash.

phkahler, 2023-07-26:
This whole thing is strange.  We only call Glib::get_charset() in one place and produce this error as a result. I'm assuming it will do something bad (like crash) somewhere is if we just go in guigtk.cpp and delete:
```
    setlocale(LC_ALL, "");
    if(!Glib::get_charset()) {
        dbp("Sorry, only UTF-8 locales are supported.");
        exit(1);
    }
    setlocale(LC_ALL, "C");
```
That check must have been added to avoid... what?

Neinei0k, 2023-08-03:
It looks like the problem only appears with flatpak version. I've tried building Solvespace and it works as expected.

I checked locale before and after `setlocale(LC_ALL, "");` call:
```
dbp(setlocale(LC_ALL, NULL));
setlocale(LC_ALL, "");
dbp(setlocale(LC_ALL, NULL));
if(!Glib::get_charset()) {
    dbp("Sorry, only UTF-8 locales are supported.");
    exit(1);
}
setlocale(LC_ALL, "C");
```
Before function call, the locale is set to `C`. `setlocale(LC_ALL, "");` sets locale to the same as specified by `LC_` environment variables, but only if environment variables contain a valid, known locale. If they contain an invalid locale or locale that is not installed, `setlocale` does not do anything. So, if `LC_TIME` is a locale that is not installed, or have some garbage value, this will result in a crash.

For some reason, flatpak did not find `ja_JP.UTF-8` locale. Because of this, `setlocale`  function call sees unknown locale `ja_JP.UTF-8` for `LC_TIME` and leaves locale at `C` which is then fails the UTF-8 test performed by`Glib::get_charset()`.

itu7, 2026-09-12:
I can confirm the same issue with version 3.2 from flathub.

```
$ /usr/bin/flatpak run --branch=stable --arch=x86_64 --command=/app/bin/solvespace --file-forwarding com.solvespace.SolveSpace
Sorry, only UTF-8 locales are supported.
```

```
$ env | egrep '(LANG|LC_)' 
LANGUAGE=en_US:en
LC_ADDRESS=de_AT.UTF-8
LC_NAME=de_AT.UTF-8
LC_MONETARY=de_AT.UTF-8
GDM_LANG=en_US.utf8
LC_PAPER=de_AT.UTF-8
LANG=en_US.UTF-8
LC_TELEPHONE=de_AT.UTF-8
LC_MEASUREMENT=de_AT.UTF-8
LC_TIME=de_AT.UTF-8
LC_NUMERIC=de_AT.UTF-8
```

Add environment variable to make Solvespace start:
```
$ /usr/bin/flatpak run --env=LC_ALL=en_US.UTF-8 --branch=stable --arch=x86_64 --command=/app/bin/solvespace --file-forwarding com.solvespace.SolveSpace
```