solvespace/solvespace #1382: Feature request: Gearing

SoerenDaenekas, 2023-07-16:
Solvespace offers a speacial kind of drawing experience using constraint based designing.

One feature I found missing though are gears (e.g. translating cogs or converting linear into rotational motion).

Creating mechanical designs like clocks, transmissions or engine attachments is currently not possible or require workarounds.

There are examples where invisible groups are used to approximate gear functionality.

For example: 
https://www.youtube.com/watch?v=W0KIwkXv030
https://www.patreon.com/posts/solvespace-rack-78622024
As seen in the downloaded project a quite complicated workaround is used to archive the desired functionality.
[image]

In this thread the idea of a cog tool is discussed:
https://github.com/solvespace/solvespace/issues/1211#issuecomment-1032030232
But I imagine that this particular idea might not cover usecases like in the video above.

My suggestion stems from my intend to design an engine-like gear switching mechanism, which it seems like solvespace in its current form is not capable to be used for.

Thanks for considering!

phkahler, 2023-07-16:
If I recall correctly, you can make an equality constraint between a line and arc in the development version. Maybe that could be used to couple the linear and angular motion.

SoerenDaenekas, 2023-07-17:
There is this tutorial: https://homehack.nl/solvespace-involute-gear/

But I discovered following downsides:
- it is quite laggy
- Rotating either cog more than one revolution results in the restraint engine breaking.
- high speed revolutions break the restraint engine
- quite a lot of work to archive this functionality
- increasing amount of teeth and the teeth distance is finicky. As seen in the tutorial it breaks if the rings are moved too fast.

SoerenDaenekas, 2023-07-18:
> If I recall correctly, you can make an equality constraint between a line and arc in the development version. Maybe that could be used to couple the linear and angular motion.

That definitly works but it comes with the downside that arcs cannot go beyond 100%. Meaning multiple revolutions aren't possible with this technique
[image]

phkahler, 2023-08-02:
> the solver does not realize it can resize the circle to make it work, as there is a degree of freedom there.

No. The degree of freedom (circle diameter) is in the previous group. Constraints are solved one group at a time, holding all previous geometry static. It might let you drag a prior group without going back to it, but that's a convenience. You do have to go back via the text window to add constraints to get zero DoF.

dgramop, 2024-04-12:
Hopefully with parameters/relation constraints, gearing will work a little better, since you can specify the rotation logic with a single constraining equation

https://github.com/solvespace/solvespace/assets/10949560/904925ef-93c7-4944-ac24-64f7a52bd55b

The math I'm using looks to be wrong but I think it demonstrates the concept

Here's the sketch
[image]

I know it look chaotic, but I have a few extra constraints to name variables that I basically use as constants. The actual logic is taken care of by a point-to-point constraint that converts the angle to an offset (of course this limits rotation to just one revolution, but if I used arcsin and cleaned up the sketch math we could totally get nicer gearing with unlimited rotation)
In this image I actually didn't use the hopefully-upcoming relation constraint, but rather named parameters.