solvespace/solvespace #1385: Feature Request: Drop last segment in a line when pressing escape

RM250J, 2023-07-26:
### System information

- **SolveSpace version:**3.1 
- **Operating system:**All 

### Expected behavior
When drawing segments (straight lines), if I haven't set the last segment endpoint that line disappears when I press Escape

### Actual behavior
When drawing segements, I click on where I want each line to end, but if I don't want to close the shape and press Escape to exit Segments the last line that I didn't want is drawn to where ever the mouse pointer was

### Additional information

ruevs, 2023-07-26:
At least the maintainer agrees with you https://github.com/solvespace/solvespace/pull/1224#issuecomment-1061832096 :-)
I got used to it over time :-D

phkahler, 2023-07-26:
@ruevs
I think in this code:

https://github.com/solvespace/solvespace/blob/f0912d42b1e6447b0857d0408fe69e7c85d9bb73/src/graphicswin.cpp#L1086

We shall add a "cancel_pending_operation()" which calls SS.UndoUndo() for those pending operations that have created an entity and are just dragging it. Maybe we inline there or maybe a new function. It's not clear to me right now if undo is appropriate for all pending operations, so adding logic in a new function seems safer.

phkahler, 2023-07-26:
@ruevs  I just couldn't let this go another day, hence the same-day PR.

closed 2023-07-27 by phkahler

RM250J, 2023-07-28:
Awesome, just compiled Solvespace and used it, really helps the flow for me. Thx!

PR #1386 (2023-07-27): Delete the partially drawn entity when ESC is pressed on pending oper…
…ations by using UndoUndo()

Fixes #1385