solvespace/solvespace #1387: CI: Ubuntu and macOS actions fail

ruevs, 2023-07-27:
### System information

- **SolveSpace version:** git head for the last four months
- **Operating system:** Github actions for Ubuntu and macOS fail in the continuous integration.

### Expected behavior

When commiting to master, or on a pull request, the Github action should build the software (and upload artifacts for master).

### Actual behavior

The CI builds are [currently broken](https://github.com/solvespace/solvespace/actions/runs/5675173296), so the only option for users to test new features is to [compile from source](https://github.com/solvespace/solvespace#via-source-code) .

### Additional information

The last successful build is [this](https://github.com/solvespace/solvespace/actions/runs/4446327039#artifacts) from March 17.

https://github.com/solvespace/solvespace/issues/739

https://github.com/solvespace/solvespace/issues/1089

Draconis-25, 2023-08-08:
Seems like ubuntu-18.04 and macos-10.15 are no longer hosted by GitHub. Switching to ubuntu-latest (at the time of writing: 22.04) and macos-latest (at the time of writing 12) works.

[GitHub hosted runners](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners#supported-runners-and-hardware-resources)

ruevs, 2023-08-09:
@Draconis-25 thank you. The CI run now passed https://github.com/solvespace/solvespace/actions/runs/5798929779

closed 2023-08-09 by ruevs