solvespace/solvespace #1390: "failed to assemble polygon to trim nurbs surface in uv space" with a simple wedge cut-out

vthriller, 2023-08-11:
### System information

- **SolveSpace version:** 3.1~70bde63c
- **Operating system:** Gentoo Linux

### Expected behavior

There is no error if extrusion does not extend all the way through the other group:

[image]

### Actual behavior

[image]

### Additional information

[wedge-diff-nurbs.slvs.gz](https://github.com/solvespace/solvespace/files/12318087/wedge-diff-nurbs.slvs.gz)

ruevs, 2023-08-11:
This is one of the "typical" [NURBS or mesh failures](https://github.com/solvespace/solvespace/issues/738).

This one appears to be in the same category as these ones:
https://github.com/solvespace/solvespace/issues/1091
https://github.com/solvespace/solvespace/issues/871

And in the same general "category" as these ones:
https://github.com/solvespace/solvespace/issues/35#issuecomment-531602706
https://github.com/solvespace/solvespace/issues/537#issuecomment-669068154
https://github.com/solvespace/solvespace/pull/594#issuecomment-625587508
https://github.com/solvespace/solvespace/issues/1091#issuecomment-898939648

ruevs, 2023-09-22:
The sad (or very interesting) part is that this model works in this version https://github.com/ruevs/solvespace/releases/tag/WindowsXPPost3.1 
Perhaps it is the same type of failure as "wrong3" https://github.com/solvespace/solvespace/issues/683#issuecomment-730602943 that [only fails in some builds](https://github.com/solvespace/solvespace/issues/683#issuecomment-731763446) probably based on whether `double` is 64bit IEEE or 80bit Intel FPU.