solvespace/solvespace #1391: unexpected "redundant constraints" for extrusion

vthriller, 2023-08-11:
### System information

- **SolveSpace version:** 3.1~70bde63c
- **Operating system:** Gentoo Linux

### Actual behavior

[image]

### Additional information

No idea how this constraint is "redundant" if, when removed, extrusion depth is no longer fixed.

This is the same model as #1390 with additional "on point" constraint (highlighted on screenshot above) that extends second extrusion to the bottom of the first one.

[wedge-diff-unexpectedly-redundant.slvs.gz](https://github.com/solvespace/solvespace/files/12318105/wedge-diff-unexpectedly-redundant.slvs.gz)

phkahler, 2023-08-11:
An extrusion has one degree of freedom - how long.  A point on point constraint reduces 3 degrees of freedom - by making the X, Y, and Z value of the point equal. That's why this results in an over constrained sketch.

Try a point-on-plane instead, as that is a 1 DoF constraint.

ruevs, 2023-08-11:
The points coincident constraint is indeed redundant, since (in 3d) it removes three degrees of freedom and the extrusion `g005-extrude` only has one.

[image]

To achieve what you want set the point to be on the "bottom" face:

[image]

Then you will have 0 DOF and no redundant constraints:

[image]

closed 2023-08-11 by ruevs