solvespace/solvespace #1392: can not launch on Win7

xyont, 2023-08-11:
### System information

- **SolveSpace version:** 3.1 stable
- **Operating system:** Windows 7 (64bit) SP1 updated

[image]

### Expected behavior

the programs can not launch

### Actual behavior

error messages
### Additional information

The application was unable to start correctly (0xc000007b). Click OK to close the application.

phkahler, 2023-08-14:
@xyont  do older versions of solvespace work on your Windows 7 ?

ruevs, 2023-08-20:
Do these versions launch?
https://github.com/ruevs/solvespace/releases/

xyont, 2023-08-20:
`solvespace_WinXPBuild_Post3_1_659215d.exe` is not working (same error messages),  some  allert notification by Windows Defender. However, version 2.0 stable is fine. 

[image]

ruevs, 2023-08-27:
Well that is just Windows defender being... Windows defender...

If you want to be extra cautious give the file (or a link to it) to https://www.virustotal.com and convince yourself that it is not a virus.

Once you do that and tell Windows defender to let it be does it work?

xyont, 2023-08-28:
i try excluded SolveSpace.exe files from Defender but still not works.

ruevs, 2023-09-06:
@xyont what error message do you get?

xyont, 2023-09-10:
After several times restarting the system, it's working now. 

Many thanks for the hints about excluded software setting in Defender.

[image]

closed 2023-09-10 by xyont