solvespace/solvespace #1393: How to orbit in 3D space? Loop when switching to In Workplane

TomasHubelbauer, 2023-08-12:
### System information

- **SolveSpace version:** Version 3.1 (3.1~70bde63c)
- **Operating system:** macOS Ventura 13.4.1

### Expected behavior

Clicking on the 3-axis gizmo in the bottom left and dragging it should take me to a 3D orbit view or tell me how to get there. It should not IMO do nothing even if the view settings are set up such that the rotation is locked or the view is locked to a 2D plane. I believe it is preferable to detect this and instruct the user for better UX.

### Actual behavior

Clicking and dragging the gizmo doesn't move the view.

### Additional information

I ran into this while following the basic tutorial here:
https://solvespace.com/bracket.pl

I have my sketch and now I am ready to extrude to the 3rd dimension. I go and click on New Group > Extrude and get this warning:

[image]

Heeding its advice I head to Sketch > In Workplane (as opposed to my currently selected Anywhere in 3D) and get this warning:

[image]

At this point I am stuck in a loop, not sure how to break out. I am a new user so I am not familiar with the nomenclature of UI of SolveSpace hence my above suggestion to detect common failure cases (like the attempt to use the Gizmo while it is locked to a 2D plane) and instruct the user how to get out of them.

TomasHubelbauer, 2023-08-12:
BTW I should probably also mention that I am a trackpad user and I don't have a mouse connected to my laptop. I tried two and three finger gestures to see if maybe I could rotate / orbit using the trackpad but did not get anywhere with that. I notice there is #68 which is open and deals with touch gestures so I assume that's why I was not able to use the trackpad to orbit.

ruevs, 2023-08-14:
https://solvespace.com/ref.pl#Graphics

When using a trackpan you can:
"rotate by Shift+right-dragging"

For "Sketch > In Workplane" to work you need to select a workplace. Workplanes are the big squares that (by default) are in the same plane as the coordinate planes (XY, YZ, XZ).

ruevs, 2023-08-14:
But on macOS it seems trackpad vs. normal (two buttons and a click able scroll wheel) vs. Magic mouse (and other Aplle oddities like one button mice) is a bit complicated.

If you like some "light" reading here are some links:
https://github.com/solvespace/solvespace/pull/1274
https://github.com/solvespace/solvespace/pull/1107
https://github.com/solvespace/solvespace/pull/1218

jkrei0, 2023-08-27:
> Clicking on the 3-axis gizmo ... It should not IMO do nothing

Clicking on the "3-axis gizmo" does actually do something - it lets you select the normals.

Although .. the behavior seems a little broken. Selecting/hovering a normal selects/colors the normals in the center of the screen, - except the `XY` (blue) normal doesn't highlight the normal in the center, and the `ZX` (green) doesn't turn red when selected (in the corner), but does highlight in the center? It's only visual - everything selected shows up as expected in the property browser.
[image]

> Clicking on the 3-axis gizmo in the bottom left and dragging it should take me to a 3D orbit view

I agree, it would be nice to be able to click and drag here to rotate, or click the arrows to align the view. I suppose there's some use for easily selecting the normals from there, but I've not needed it.