solvespace/solvespace #1394: Main window hijacks the Cmd+C keyboard shortcut when the About modal is focused

TomasHubelbauer, 2023-08-12:
### System information

- **SolveSpace version:** Version 3.1 (3.1~70bde63c)
- **Operating system:** macOS Ventura

### Expected behavior

When on macOS and going to SolveSpace > About SolveSpace to be able to copy out the SolveSpace version in order to file for example a GitHub issue, when I select the SolveSpace version in the dialog window and press Cmd+C I see this modal:

[image]

Clearly the main window handles this shortcut combo despite the utility window being open and focused.
I am able to right-click and select Copy in the context menu to copy the version.

### Actual behavior

The About SolveSpace window should not bubble the Cmd+C shortcut to the main window.

### Additional information

This is a super minor thing but I thought it was still worth mentioning.

jkrei0, 2023-08-26:
I'd like to add, that on windows, it isn't possible to select the text in the about dialog at all.

There is also `Help > Go to GitHub Commit` (not sure if it's different on macOS) which opens the commit in the browser, and you can copy from there.

ruevs, 2023-09-02:
> I'd like to add, that on windows, it isn't possible to select the text in the about dialog at all.

On Windows pressing `CTRL+C` while the About dialog is open copies all the text from it to the clipboard. This is true for any standard Windows message box and thus for all message boxes that SolveSpace shows.:

```
---------------------------
SolveSpace - Message
---------------------------
This is SolveSpace version 3.1~659215d8.

For more information, see http://solvespace.com/

SolveSpace is free software: you are free to modify
and/or redistribute it under the terms of the GNU
General Public License (GPL) version 3 or later.

There is NO WARRANTY, to the extent permitted by
law. For details, visit http://gnu.org/licenses/

© 2008-2023 Jonathan Westhues and other authors.

---------------------------
OK   
---------------------------
```

phkahler, 2023-09-02:
> On Windows pressing `CTRL+C` while the About dialog is open copies all the text from it to the clipboard.

I didn't know that, but it's also way more inconvenient than selecting just the text of the version number ;-)