solvespace/solvespace #1397: CTRL+Shift+S should SaveAs, not save.

greenfox1505, 2023-08-20:
### System information

- **SolveSpace version:** 
- **Operating system:** 

### Expected behavior

"Save As"

### Actual behavior

"Save"

### Additional information

Most applications bind CTRL+Shift+S as "save as" and CTLR+S as "save". Honestly, I'd be hard pressed to name one that didn't.

ruevs, 2023-08-20:
Already implemented in the master branch https://github.com/solvespace/solvespace/commit/7ccb0ffac46e34abdef038ee1dacab3b21adeb93

Please try an edge build and report if it does not work for you:
https://github.com/solvespace/solvespace#via-automated-edge-builds

ruevs, 2023-09-06:
@greenfox1505 did you try it?

phkahler, 2023-09-12:
I verified this is working on Windows with an edge build.

closed 2023-09-12 by phkahler