solvespace/solvespace #1398: unstable "failed to assemble polygon to trim nurbs surface in uv space"

vthriller, 2023-08-25:
### System information

- * **SolveSpace version:** 3.1~70bde63c
- * **Operating system:** Gentoo Linux

### Actual behavior

...depends on extrusion depth.

[image]

### Additional information

[nurbs-2.slvs.gz](https://github.com/solvespace/solvespace/files/12435894/nurbs-2.slvs.gz)

ruevs, 2023-09-06:
For cross reference https://github.com/solvespace/solvespace/issues/1390.

phkahler, 2023-09-10:
Some observations: This happens intermittently for all 3 boolean operations (differnce, union, intersection). It also happens if we replace the arc in the first sketch with 3 line segments approximating the angles at the attachment points. It does not have a problem if the final extrusion length does not extend past that large flat surface.

phkahler, 2023-09-10:
Update by replacing the arc with  2 lines and adjusting the extrusion to show the issue I get this:
[image]

Then dragging the sharp corner around the problem goes away if that triangular part is above or to the right of the area in quesion. I believe this is the same as issue #1367 which I just updated with a picture and some ideas on what's failing.

vthriller, 2025-11-28:
Following https://github.com/solvespace/solvespace/issues/1390#issuecomment-3590653146 I also tested file from here, and it too is only reproducible with OpenMP build.